sandeepmistry/arduino-LoRa #448: LoRa.rssi() always returns 0

sensorhost, 2021-03-04:
I have updated LoRa.cpp and LoRa.h so that they include the rssi function.
I want to be able to monitor the rssi over time.

My have quite a few other projects sending and receiving LoRa packets that all work well on this hardware, however the following always shows rssi as zero.
I don't want packetRssi, I just want rssi.

Example code as follows;
```
#include <Arduino.h>
#include <SPI.h>
#include <LoRa.h>

#define SX1276_DIO0 5
#define SX1276_RESET 9
#define SX1276_CS 10
#define SX1276_MISO 11
#define SX1276_MOSI 12
#define SX1276_SCK 13

#define LORA_FREQ       919450000
#define LORA_SYNCWORD   0xF3
#define LORA_BANDWIDTH  250E3
#define LORA_CODERATE   5
#define LORA_SPREADF    10

//===========================================================
void setup() {

  Serial.begin(115200);
  delay(5000);                    // Delay while USB enumerates

  LoRa.setPins(SX1276_CS, SX1276_RESET, SX1276_DIO0);
  if (!LoRa.begin(LORA_FREQ)) { 
    Serial.println(F("Failed!"));
  }

  Serial.println(F("\r\n\r\nRunning"));

}

//===========================================================
void loop() {
  int r = LoRa.rssi();
  if (r > -157){
    Serial.println(r);
  }
}
```

Even if I just continually print out LoRa.rssi(), it is always the same. I have another lora receiver on my workbench (using the same library) and it is receiving lora packets every minute ok, but the -157 never changes on this code.

Can anyone provide some help?

IoTThinks, 2021-03-04:
To add LoRa.receive() in the last line of setup().

sensorhost, 2021-03-04:
Thanks @IoTThinks, I also just found that on #429
Have been searching for it for a few hours.
Typical, I find the answer 3 minutes after posting an issue.

This will also be used for crude CAD, do I need to do anything special to change from receive mode to transmit?

IoTThinks, 2021-03-04:
With this library, you only need to set receive() after sending to continue receiving.
When you send a packet, the library will set the mode to TX before sending.

IoTThinks, 2021-03-04:
CAD is more accurate detection of LoRa packets.
rssi() will sense non-LoRa signals, too.

sensorhost, 2021-03-04:
Thanks, yes I would prefer to use CAD, however it seems that's not quite ready in this library yet.
I don't mind about the non-lora signals, as I don't want to transmit if there is any other interferer, so I can delay until the spectrum is free so to speak.

Do you know how Wideband is the rssi? Does it work over the specified bandwidth at my centre frequency or does it work over the entire receive spectrum of the module?

IoTThinks, 2021-03-04:
> Do you know how Wideband is the rssi? Does it work over the specified bandwidth at my centre frequency or does it work over the entire receive spectrum of the module?

I have no idea. I believe it is not mentioned in Semtech specification.

sensorhost, 2021-03-04:
@IoTThinks thanks for all your help and super quick responses.

closed 2021-03-04 by sensorhost