sandeepmistry/arduino-LoRa #449: Very high RSSI values even at tiny distance.

Haxk20, 2021-03-15:
I have recently soldered my SX1276 modules to my custom board which has U.FL connector with U.FL to SMA cable connected to it and SMA has 5dB antenna connected to it.
SX1276 is connected to ESP32 using its 3.3V line as input voltage.
SX1276 is configured to use 868Mhz.
Custom board gerber files:  [Here](https://github.com/sandeepmistry/arduino-LoRa/files/6143985/Ionium.zip)
Source code of the project: [Here](https://github.com/Ionium-Project/Ionium)
Link to antenna im using: [Here](https://www.ebay.com/itm/868MHz-5dBi-Wireless-Antenna-Router-Antenna-RP-SMA-Female-to-IPX-1-13-Cable/133459988298?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649)
Modules im using: [Here](https://www.ebay.com/itm/RFM95-SX1276-LoRa-Long-Range-Wireless-Transceiver-Module-868Mhz/324115137004?hash=item4b76c481ec:g:nvUAAOSwHlxee93H)

Now for the RSSI values.

Whatever i did in the configuration i just could not get a better signal from it.
I tried disconnecting the IPX to SMA cable and antenna and attaching a home made 868Mhz antenna as according to specs. Didnt improve. Actually became worse.
At literally antenna touching distance the best RSSI value i got was around -40. Weirdly enough when i sent message from the second device the RSSI values they were better. -45dB compared to -38dB. This was true for every packet i sent.
At around 1 meter the RSSI dropped to around -65dB and at 5 meters to around -105dB.

Surprisingly i did a small range test where i left one device in my room and went outside about 100 meters. So there were around 4 houses in the way of the signal. The packet sent from 80 meters arrived with RSSI of -109dB and the one at 100 meters failed to arrive at all (probably due to the houses blocking the way).

I have not yet tried powering the modules from external power (Waiting for 5V to 3.3V converter board)
On the board i have just soldered the right and middle pin of where the 2 way switch should go (Top left corner of the board) together for now as the switches havent yet arrived.

I did try and use the Receiver and Transmitter example codes. Same exact results. So i think we can say the code is fine.

I think that covers the entire issue at hand.
Im more than glad to get any help regarding this as 100 meter range on LoRa modules is well lets just say not great when the project hopes to get at least 1km :smile:

IoTThinks, 2021-03-16:
The range is only valid if there is no obstacle in between.
Show the physical / real photo of your full kits with antenna. 

> At around 1 meter the RSSI dropped to around -65dB and at 5 meters to around -105dB.
The signal at 5m (without obstacles?) is very bad.

Haxk20, 2021-03-16:
> The range is only valid if there is no obstacle in between.
> Show the physical / real photo of your full kits with antenna. 
> 

Will get some photos. And post them later today.

> > At around 1 meter the RSSI dropped to around -65dB and at 5 meters to around -105dB.
> 
> The signal at 5m (without obstacles?) is very bad. 

Completely clear line of sight.

> Either not enough current or bad LoRa chip position or wrongly tuned antenna.
All of them are possible. The antenna isnt exactly expensive and i saw the same antenna selling for 433Mhz too. And even 915Mhz.

Bad chip position idk about that. But well you will see when i post the pictures of the real hardware later today.

I did wonder about the current but after some duckduckgo searching it seems ESP32 provides like 300mA at 3.3V line.

Haxk20, 2021-03-16:
> The range is only valid if there is no obstacle in between.
> Show the physical / real photo of your full kits with antenna. 
> 
> > At around 1 meter the RSSI dropped to around -65dB and at 5 meters to around -105dB.
> 
> The signal at 5m (without obstacles?) is very bad. 
> Either not enough current or bad LoRa chip position or wrongly tuned antenna.

[image]
[image]
[image]

Here are the images as requested. I added also the empty pcb from front and back for reference.

IoTThinks, 2021-03-16:
From your photo, the impedance of the trace from RFM module to the ipex is very likely wrong.

It is quite tricky to make it right.

Is this your custom PCB?

IoTThinks, 2021-03-16:
From my experience, without proper equipment, the ipex should be as near as possible.

The best is directly put into antenna pin of RFM module.

Haxk20, 2021-03-16:
> From my experience, without proper equipment, the ipex should be as near as possible to the antenna pin of RFM module.
> 
> The best is directly put into antenna pin of RFM module.
The trace is is really tiny but i will keep it in mind to make it next to the antenna pin on next revision of the board.

Tho when connecting the normal antenna directly to the antenna pin the results didnt improve. They stayed the same or even became worse. Is this expected or does this mean the issue is also somewhere else ?

IoTThinks, 2021-03-17:
> Tho when connecting the normal antenna directly to the antenna pin the results didnt improve. They stayed the same or even became worse. Is this expected or does this mean the issue is also somewhere else ?

Then try another antenna or use a bronze wire directly to the antenna hole of the RFM modules.
With no obstacle, the RSSI of 868-916Mhz should be around -70.

For 868Mhz, the bronze antenna should be around 299792458 / (868000000 * 4) * 100 * 0.95 = 8.2cm.

You can display the SNR to see if the SNR is ok. 
For 5m, it should be around 10-14.

IoTThinks, 2021-03-18:
Check if you are using wrong RP-SMA antenna and SMA connector?

SMA and RP-SMA are not compatible.

[image]

Haxk20, 2021-03-18:
> Check if you are using wrong RP-SMA antenna and SMA connector?
> 
> SMA and RP-SMA are not compatible.
> 
> https://forum.digikey.com/t/sma-vs-rp-sma-what-is-the-difference/550
> 
> [image]

Im certain im using the correct connector as it came with the antenna.

Haxk20, 2021-03-18:
> Then try another antenna or use a bronze wire directly to the antenna hole of the RFM modules.
> With no obstacle, the RSSI of 868-916Mhz should be around -70.
>
> For 868Mhz, the bronze antenna should [...]

Will try when i get time.

Haxk20, 2021-04-08:
> Then try another antenna or use a bronze wire directly to the antenna hole of the RFM modules.
> With no obstacle, the RSSI of 868-916Mhz should be around -70.
>
> For 868Mhz, the bronze antenna should [...]

I guess a small update is in order. I found some tiny amount of time and checked the RSSI with copper wire attached directly to the antenna hole. At aprox 1Meter the RSSI was around 50-58.
I sadly didnt get the time to edit the code and add SNR. Will try to do as soon as possible tho.

Haxk20, 2021-04-08:
So the RSSI definetly improved a tiny bit but nothing to write world about.

IoTThinks, 2021-04-08:
At 1m, the rssi -50 is good already.

Haxk20, 2021-04-08:
> At 1m, the rssi -50 is good already.
> 
> Try again at 5m with copper wire.

Will do. Tho cant even aproximate when. Uni and work and just real life.

IoTThinks, 2021-04-09:
You should not complain about your busy life.
We are helping you at good will. 
It doesn't imply that we have more free time than you.

Usually, students have more free time then any one else.
BTW, your are not required to perform the tests. The tests benefit you, not us. :)

Haxk20, 2021-04-10:
Uhmm i didnt mean it in any bad way. But sure.