sandeepmistry/arduino-LoRa #450: Compatibility with the RadioHead Library

sergerold, 2021-03-23:
Hi

I am working on a project using: 

1. a RFM95 device on my Arduino using this arduino-LoRa library;
2. a RFM95 device on my RPi using the [RadioHead](https://www.airspayce.com/mikem/arduino/RadioHead/) library.

I know that both of these LoRa devices are working properly.

I am struggling to get these two devices to communicate using these two different libraries. When the RPi sends a message using the RadioHead library, the Arduino LoRa receives a package but it seems to be garbage.

Has anyone had success using the arduino-LoRa library on one LoRa device with the RadioHead library on another device? I assume that LoRa specifies a common data format with the libraries providing different ways to access this common data format?

I believe the syncwords are the same under both libraries (0x12).

Thanks

I've pasted below the code for each device:

```
**RadioHead library**

    #include <pigpio.h>
    #include <stdio.h>
    #include <signal.h>
    #include <unistd.h>
        
    #include <iostream>
        
    #include <RH_RF95.h>

    //Function Definitions
    void sig_handler(int sig);
    
    //Pin Definitions
    #define RFM95_CS_PIN 7
    #define RFM95_IRQ_PIN 16
    #define RFM95_LED 13
    
    //RFM95 Configuration
    #define RFM95_FREQUENCY 915.00
    #define RFM95_TXPOWER 20
    
    // Singleton instance of the radio driver
    RH_RF95 rf95(RFM95_CS_PIN, RFM95_IRQ_PIN);
    
    //Flag for Ctrl-C
    int flag = 0;
    
    //Main Function
    int main (int argc, const char* argv[] )
    {
      if (gpioInitialise()<0)
      {
        printf( "\n\nRPI rf95_client startup Failed.\n" );
        return 1;
      }
    
      gpioSetSignalFunc(2, sig_handler); //2 is SIGINT. Ctrl+C will cause signal.
    
      if (!rf95.init())
      {
        printf( "\n\nRF95 driver failed to initialize.\n\n" );
        return 1;
      }
    
    
      /* Begin Manager/Driver settings code */
    
      rf95.setTxPower(RFM95_TXPOWER, false);
      rf95.setSpreadingFactor(12);
      rf95.setSignalBandwidth(125E3);
      rf95.setCodingRate4(5);
      rf95.setFrequency(RFM95_FREQUENCY);
      rf95.setPayloadCRC(true);
      /* End Manager/Driver settings code */
    
      /* Begin Datagram Client Code */
      while(!flag)
      {
        Serial.println("Sending to rf95_server");
        // Send a message to rf95_server
    #ifdef RFM95_LED
        gpioWrite(RFM95_LED, PI_ON);
    #endif
        uint8_t data[] = "Hello World!";
        rf95.send(data, sizeof(data));
    
        rf95.waitPacketSent();
    #ifdef RFM95_LED
        gpioWrite(RFM95_LED, PI_OFF);
    #endif
    
        gpioDelay(400000);
      }
      printf( "\nrf95_client Tester Ending\n" );
      gpioTerminate();
      return 0;
    }
    
    void sig_handler(int sig)
    {
      flag=1;
    }
```
 
```
**//arduino-LoRa library**
#include <LoRa.h>

// Radio settings
#define FREQ 915E6
#define TX_PWR 20
#define SPREAD_FACTOR 12
#define BANDWIDTH 125E3 // lower bandwiths don't seem to work...
#define CODING_DENOM 5

// PINS //
#define CS_M0  8
#define RST_M0 4
#define INT_M0 3

LoRaClass &remoteStationRadio = LoRa; // alias

char RX_BUFFER[255]; // add one space for null terminator
char TX_BUFFER[255]; // add one space for null terminator

void setupRadio()
{
  remoteStationRadio.setPins( CS_M0, RST_M0, INT_M0);
  while (!remoteStationRadio.begin(FREQ)) {
      Serial.println("Starting remoteStationRadio failed!");
  }
  
  remoteStationRadio.setTxPower(TX_PWR);
  remoteStationRadio.setSpreadingFactor(SPREAD_FACTOR);
  remoteStationRadio.setSignalBandwidth(BANDWIDTH);
  remoteStationRadio.setCodingRate4(CODING_DENOM);
  remoteStationRadio.enableCrc();
}

bool awaitRxForSeconds(uint32_t seconds_to_wait, char* rx_str, LoRaClass& radio)
{
    rx_str[0] = '\0'; // null terminate
    unsigned long t_start = millis(); // start timer
    while ( (millis() - t_start) <= seconds_to_wait)
    {
        int packetSize = radio.parsePacket();
        if (packetSize)
        {
          Serial.print("Packet received:");
          // read packet
          size_t rx_str_len = 0;
          while (radio.available()) // extract all of the data, byte by byte
          {
            rx_str[rx_str_len] = static_cast<char> ( radio.read() ), rx_str[++rx_str_len] = '\0'; // insert each extracted byte into the buffer and append null symbol
          }
          Serial.println(rx_str);
          return true;
        }
    }
    return false;
}

void setup() {
  Serial.begin(9600);
  setupRadio();

}

void loop() {
  awaitRxForSeconds(5, RX_BUFFER, remoteStationRadio);

}
```

IoTThinks, 2021-03-24:
Try to use same library in both sender and receiver first.

I use this library and RadioLib. Works together fine.

closed 2021-04-22 by IoTThinks

ilomon10, 2021-06-15:
@IoTThinks Can you make some example to use with Radiohead please.

IoTThinks, 2021-06-15:
Try the codes in example folders of this library.

All LoRa libraries make the same register calls to LoRa SX127x.