sandeepmistry/arduino-LoRa #451: Cant transmit, send packets -  loop stops while transmitting 1st packet

michalmutka, 2021-04-01:
I am using the example code for transmitting, but loop stops and waits for 1st packet to be sent, but it does not send. I got SX1276RFIAS and arduino UNO board wired as in description.

Any tips?
[image]

IoTThinks, 2021-04-01:
Remove your code of flame sensor and try again.

michalmutka, 2021-04-01:
> Why 1690E6?
> Highly doubt if LoRa supports over 950Mhz.

Sorry, thats a missclick. Even when I put 870 MHz or any other freq, it is the same.

michalmutka, 2021-04-01:
> Why 1690E6?
> Highly doubt if LoRa supports over 950Mhz.

Already uploaded new photo with fixed code.

IoTThinks, 2021-04-01:
Are you sure if the chip is not 433Mhz?

Remove your code of flame sensor  first.

Run the default example.

michalmutka, 2021-04-01:
I tried, it is still the same, even without the flame sensor code.
Yes I am sure, the module SX1276RF1IAS supports two antenas -  one for 169MHz and one for 868 MHz. Neither of these frequencies work..
[image]

IoTThinks, 2021-04-01:
Check this? https://github.com/sandeepmistry/arduino-LoRa/issues/47

IoTThinks, 2021-04-01:
You can try to use LoRa.endPacket(true) to make sure the LoRa doesn't stuck at endPacket.
However, it seems the issue of the board not the library.

closed 2021-04-22 by IoTThinks