sandeepmistry/arduino-LoRa #456: How can I access packets in an array?

farbod001, 2021-04-07:
Hi, 
Thanks for this amazing library.
I just need to know if there's a way to be able to compare the received packets with a bye or a word or anything on the Receiver?

Example:
if (packet == "Message")
{
Do something
}
or even : 
if (packet[0] == EEPROM[0] && packet[2]==EEPROM[2])
{
Do something
}
I looked very hard into the code to find a function like this. 
It appears you have LoRa.peek() and LoRa.read() to read message bytes. But I couldn't compare their contents to anything. 
Do you mind helping me a bit?

IoTThinks, 2021-04-08:
Check the example codes in the library.

farbod001, 2021-04-08:
Great! I found the answer to my question in "LoRa Duplex communication with Sync Word" example. 
By the way, this is a very good library. I wasted a lot of time on another library and I was losing %80 of my messages. In this one, using the same hardware as before, I lose 0 messages. You guys are amazing keep it up!

closed 2021-04-08 by IoTThinks