sandeepmistry/arduino-LoRa #458: LoRaReceiverCallback.ino fails after a number of runs when the example ISR is expanded

Erik84750, 2021-04-12:
When code is added to the void onRceive(int packetsize) function this causes reception to halt after a number of runs.
Incoming data consists of 14 bytes: two characters, 1 integer (2byte) and 2 floats (4byte).
These need to be parsed: see code below.
In the main loop there obviously is also code that needs to be executed.

Tried but failed to help:
1. adding Serial.flush() at the end of this routine
2. removed all but one Serial.print comment outputs to serial monitor
3. increased serial monitor baud rate from 9600 to 115200
4. added a flag to halt main loop code execution when receive callback is activated (ie DIO0 is pulsed high (50usec).
Nothing of the above worked.
Removing all delay() from the main loop does seem to help; but how helpful is this when this main loop needs to execute lots of code?

I refer also to issue #341: the implementation of DIO0 interrupt call; there it is explained how this library functions to get receive callback works.

How could this issue be best resolved?

`
void onReceive(int packetSize) {

  if (packetSize == 0) {
    return;
  }
  doRead = true;
  //  incomingPacketSize = packetSize;
  // Read 2 bytes into a
  uint16_t a;
  char LoRaStringA[12];
  for (int i = 0; i < sizeof(a); i++) {
    *(((uint8_t*)&a) + i) = (uint8_t)LoRa.read();
    //     LoRaString = a+(char)LoRa.read();
    // itoa(a, LoRaStringA, 12);
  }
  char a1 = a;

  // Read 2 bytes into b
  uint16_t b;
  char LoRaStringB[12];
  for (int i = 0; i < sizeof(a); i++) {
    *(((uint8_t*)&b) + i) = (uint8_t)LoRa.read();
    //     LoRaString = b+(char)LoRa.read();
    // itoa(b, LoRaStringB, 12);
  }
  char b1 = b;

  // Read 2 bytes into counter
  uint16_t counter;
  for (int i = 0; i < sizeof(counter); i++) {
    *(((uint8_t*)&counter) + i) = (uint8_t)LoRa.read();
  }
  // Read 4 bytes into x
  float x;
  for (int i = 0; i < sizeof(x); i++) {
    *(((uint8_t*)&x) + i) = (uint8_t)LoRa.read();
  }
  // Read 4 bytes into z
  float z;
  for (int i = 0; i < sizeof(z); i++) {
    *(((uint8_t*)&z) + i) = (uint8_t)LoRa.read();
  }
  // Display read values
  /*
  Serial.print("a = ");
  Serial.print(a1);
  Serial.print("  b = ");
  Serial.print(b1);
  */
  Serial.print("  counter = ");
  Serial.println(counter);
  /*
  Serial.print("  x = ");
  Serial.print(x);
  Serial.print("  z = ");
  Serial.print(z);
  Serial.print("  with RSSI ");
  Serial.println(LoRa.packetRssi());
  Serial.flush();
  */
  doRead = false;    // Set flag back to false so next read will happen only after next ISR event
}

`

Erik84750, 2021-04-12:
SOLVED,.. I think: after discovering nanosecond spikes on the reset line, the scope was put on the 3V rail: same. Changed power supply; spikes gone.
So far so good..(couple of hours).

Would there be a limit to what can be done within the void onReceive(int packetSize) ?
And to what can be done in the main loop?

IoTThinks, 2021-04-12:
Then likely your board doesn't supply enough power.
Arduino Nano?

Erik84750, 2021-04-12:
Pro Mini 3.3V 8MHz supplied by lab PSU 3.3V 5A limit.

Erik84750, 2021-04-12:
Using the original LoRaReceiverCallback.ino program it lasted 52 runs before that too stopped.
Is this then a hardware issue?

IoTThinks, 2021-04-13:
You can try a Uno or esp32?

My esp32 can listen for packets in months.

Erik84750, 2021-04-13:
Must indeed be hardware issue; for hours being run now on 2x UNO with Dragino LoRa shield ( [https://wiki.dragino.com/index.php?title=Lora_Shield](url)  ), no issues.

Erik84750, 2021-04-13:
Replaced the hardware of my initial test (Pro Mini 8MHz and LoRa module) and working perfectly for hours now.
So this definitely was a hardware issue. Great library, callback provision is a fantastic add-on, thank you.

IoTThinks, 2021-04-13:
So you change to another board of Pro Mini? Or Uno board?

IoTThinks, 2021-04-13:
> Must indeed be hardware issue; for hours being run now on 2x UNO with Dragino LoRa shield ( [https://wiki.dragino.com/index.php?title=Lora_Shield](url)  ), no issues.

Glad it works.

closed 2021-04-22 by IoTThinks