sandeepmistry/arduino-LoRa #459: ESP32: float value in ISR (Guru Meditation Error: Core  1 panic'ed (Coprocessor exception))

sacculus, 2021-04-20:
Hi
I'm write sketch for esp32 board with event driven model when send/receive.
I got exception when have tried print SNR & frequency error in onReceive callback like this:

`Serial.printf("% 10d: LoRa: RSSI: %d, SNR: %0.2f, Frequency error: %d\n", millis(), LoRa.packetRssi(), LoRa.packetSnr(), LoRa.packetFrequencyError());`

Exception:
`Guru Meditation Error: Core  1 panic'ed (Coprocessor exception)
Core 1 register dump:
PC      : 0x400d0fee  PS      : 0x00060a31  A0      : 0x800d17c2  A1      : 0x3ffbea50  
A2      : 0x0000153f  A3      : 0x00000001  A4      : 0x3ffbffc4  A5      : 0x00000000  
A6      : 0xffffffb7  A7      : 0x0000002f  A8      : 0x800d0fee  A9      : 0x3ffbeaf0  
A10     : 0x41080000  A11     : 0x0000005b  A12     : 0xc87536dc  A13     : 0x3ffc0794  
A14     : 0x3ffbeaac  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x00000004  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff4  
Core 1 was running in ISR context:
EPC1    : 0x400d0fee  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400841af

ELF file SHA256: 0000000000000000

Backtrace: 0x400d0fee:0x3ffbea50 0x400d17bf:0x3ffbeb10 0x400d17bf:0x3ffbeb50 0x400811e6:0x3ffbeb70 0x40081231:0x3ffbeb90 0x40083f95:0x3ffbebb0 0x400d2804:0x3ffb1fb0 0x4008689d:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8`

Decoded exception:

`PC: 0x400d0fee: onReceive(int) at /data/Projects/Arduino/sketch/SandeepMistry/lora_client_32/lora_client_32.ino line 145
EXCVADDR: 0x00000000

Decoding stack results
0x400d0fee: onReceive(int) at /data/Projects/Arduino/sketch/SandeepMistry/lora_client_32/lora_client_32.ino line 145
0x400d17bf: LoRaClass::handleDio0Rise() at /data/Projects/Arduino/libraries/LoRa/src/LoRa.cpp line 712
0x400d17bf: LoRaClass::handleDio0Rise() at /data/Projects/Arduino/libraries/LoRa/src/LoRa.cpp line 712
0x400811e6: LoRaClass::onDio0Rise() at /data/Projects/Arduino/libraries/LoRa/src/LoRa.cpp line 751
0x40081231: __onPinInterrupt at /home/mike/.arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/esp32-hal-gpio.c line 220
0x400d2804: loopTask(void*) at /home/mike/.arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/main.cpp line 24
0x4008689d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143`

Googling led me to this [issue](https://github.com/espressif/arduino-esp32/issues/3661#issuecomment-578224260).
I replased float to double in library source for test and error has gone.
What I've changed:

`diff -urN LoRa.ORIG/src/LoRa.cpp LoRa/src/LoRa.cpp
--- LoRa.ORIG/src/LoRa.cpp	2020-12-05 17:38:41.000000000 +0200
+++ LoRa/src/LoRa.cpp	2021-04-20 22:51:42.099699743 +0300
@@ -266,7 +266,7 @@
   return (readRegister(REG_PKT_RSSI_VALUE) - (_frequency < RF_MID_BAND_THRESHOLD ? RSSI_OFFSET_LF_PORT : RSSI_OFFSET_HF_PORT));
 }
 
-float LoRaClass::packetSnr()
+double LoRaClass::packetSnr()
 {
   return ((int8_t)readRegister(REG_PKT_SNR_VALUE)) * 0.25;
 }
@@ -284,8 +284,8 @@
      freqError -= 524288; // B1000'0000'0000'0000'0000
   }
 
-  const float fXtal = 32E6; // FXOSC: crystal oscillator (XTAL) frequency (2.5. Chip Specification, p. 14)
-  const float fError = ((static_cast<float>(freqError) * (1L << 24)) / fXtal) * (getSignalBandwidth() / 500000.0f); // p. 37
+  const double fXtal = 32E6; // FXOSC: crystal oscillator (XTAL) frequency (2.5. Chip Specification, p. 14)
+  const double fError = ((static_cast<double>(freqError) * (1L << 24)) / fXtal) * (getSignalBandwidth() / 500000.0); // p. 37
 
   return static_cast<long>(fError);
 }
diff -urN LoRa.ORIG/src/LoRa.h LoRa/src/LoRa.h
--- LoRa.ORIG/src/LoRa.h	2020-12-05 17:38:41.000000000 +0200
+++ LoRa/src/LoRa.h	2021-04-20 22:49:43.286220332 +0300
@@ -42,7 +42,7 @@
 
   int parsePacket(int size = 0);
   int packetRssi();
-  float packetSnr();
+  double packetSnr();
   long packetFrequencyError();
 
   int rssi();
`

IoTThinks, 2021-04-21:
I hit the same issue for esp32.
When I put the packetSNR outside the ISR, the issue was gone.

Wonder how it affects Arduino boards?

Could you create a pull request for this?

sacculus, 2021-04-21:
> I hit the same issue for esp32.
> When I put the packetSNR outside the ISR, the issue was gone.

Not only packetSNR, packetFrequencyError leds this exception too.

IoTThinks, 2021-04-21:
I remember packetSNR works fine in ISR for my esp32.

Only packetFrequencyError hits issue for me in ISR.

italocjs, 2021-10-01:
Any bypass or fix for this beside using double? i'm having the exact same issue...

sacculus, 2021-10-03:
> Any bypass or fix for this beside using double? i'm having the exact same issue...

Hi
Some info is there in this [thread](https://esp32.com/viewtopic.php?t=1292), take a look on this [post](https://esp32.com/viewtopic.php?t=1292#p6070).

italocjs, 2021-10-04:
> > Any bypass or fix for this beside using double? i'm having the exact same issue...
> 
> Hi Some info is there in this [thread](https://esp32.com/viewtopic.php?t=1292), take a look on this [post](https://esp32.com/viewtopic.php?t=1292#p6070).

Thanks, but i ended working with int (old float * 100), my application needed to be as optimized as possible duo a massive amount of requests per second (the interrupt was being called by a CANBUS event)

huexpub, 2021-12-16:
you must use xqueue  with get isr interrupt