sandeepmistry/arduino-LoRa #462: Why does not loop Txdone?

thanapol199717, 2021-04-22:
Hi everyone, I have tried these two code snippets in order to send and receive data between them. 

(https://github.com/sandeepmistry/arduino-LoRa/blob/master/examples/LoRaSimpleGateway/LoRaSimpleGateway.ino)
(https://github.com/sandeepmistry/arduino-LoRa/blob/master/examples/LoRaSimpleNode/LoRaSimpleNode.ino)

I think they both refuse to enter receiver mode as they keep on sending.

I'm using an Arduino nano with a SX1276.

IoTThinks, 2021-04-22:
Try this https://github.com/sandeepmistry/arduino-LoRa/blob/master/examples/LoRaSenderNonBlockingCallback/LoRaSenderNonBlockingCallback.ino

And check if the TxDone is trigger.

thanapol199717, 2021-04-22:
[image]

I think it's not a trigger.

thanapol199717, 2021-04-22:
[image]

[image]

[image]

[image]

[image]

This  is my register.

IoTThinks, 2021-04-22:
> [image]
> 
> I think it's not a trigger.

Frankly speaking, I never use TxDone event. :dagger: 
From your image, TxDone is printed once. => Hence it is called once?

thanapol199717, 2021-04-22:
Yes i don't know why.

IoTThinks, 2021-04-22:
I will try with my esp32 + SX1276 this evening and see.

thanapol199717, 2021-04-22:
Okay, thank you very much, I have to try to get a 3.3v supply for the SX1276 and I can use the Uno?

IoTThinks, 2021-04-22:
Uno is better. Sx127x can use 3.3v supply from Uno.

thanapol199717, 2021-04-22:
Okay, I'll try and tell you the results

thanapol199717, 2021-04-22:
I have tried not enough from supply. Thank you so much. Do you have an example of a Raspberrypi Gateway that can reply to Arduino?
The same is true for the example I sent. But I want it to be Raspberrypi and Arduino to control my device.

Thank you for giving me new knowledge.

IoTThinks, 2021-04-22:
So the TXDone works fine with external power on Arduino Nano?

thanapol199717, 2021-04-22:
Yeah, I think so. I let sx1276 get 3.3v power from Uno, but the control is from Nano.

IoTThinks, 2021-04-22:
Ok solved.
For the rest, you need to do some Google on your own first.

thanapol199717, 2021-04-22:
Thank you very much.

closed 2021-04-23 by IoTThinks