sandeepmistry/arduino-LoRa #463: Single packet is shown as received twice

gjt211, 2021-04-30:
Board: MKRZero
LoRa Module: NiceRF SX1276-915Mhz

I have a custom board and have the following code that works except that is shows the same packet twice.
I have reduced the code to the bare minimum needed for a test.
```
#include <SPI.h>
#include <LoRa.h>

const int P_LORA_NSS = 7;
const int P_LORA_RESET = 6;
const int P_LORA_DIO0 = 10;

struct __attribute__((packed)) SENSOR_DATA11b {
  uint8_t smodel;             // 1
  uint16_t id;                // 2
  int16_t var1 = 0;           // 2
  int16_t var2 = 0;           // 2
  int16_t var3 = 0;           // 2
  int16_t crc;                // 2
};                            // 11 bytes total

SENSOR_DATA11b sData11b;

//=============================================================
void setup() {
  
  pinMode(32,OUTPUT);
  digitalWrite(32,HIGH);
  pinMode(5,OUTPUT);
  digitalWrite(5,LOW);         // Ensure mPCIE power supply is off
  pinMode(0,OUTPUT);
  digitalWrite(0,LOW);
  
  Serial.begin(115200);
  delay(2000);
  
  lora_begin();

  Serial.print(F("READY_\r"));
  Serial.println();

}

//=============================================================
void loop() {
  int packetSize = LoRa.parsePacket();
  if (packetSize == 11){
    LoRa.readBytes((uint8_t *)&sData11b, packetSize);
    Print11b(LoRa.packetRssi());
  }
}

// =============================================================
void lora_begin(){
  Serial.println(F("LoRa start"));
  LoRa.setPins(P_LORA_NSS, P_LORA_RESET, P_LORA_DIO0);
  if (!LoRa.begin(919450000)) {
    Serial.println(F("LoRa modem failure - System halted"));
    for(;;) { 
      // do nothing...
      delay(100);
    } 
  }
  LoRa.setSyncWord(0x12);
  LoRa.setSignalBandwidth(250E3); // defaults to 125E3
  LoRa.setCodingRate4(5);   // denominator of the coding rate, defaults to 5
  LoRa.setSpreadingFactor(10);
  Serial.println(F("LoRa ready"));
}

// =============================================================
void Print11b(int rssi){
  Serial.print(F(" => "));
  Serial.print(sData11b.smodel);
  Serial.print(F(","));
  Serial.print(sData11b.id);
  Serial.print(F(","));
  Serial.print(sData11b.var1);
  Serial.print(F(","));
  Serial.print(sData11b.var2);
  Serial.print(F(","));
  Serial.print(sData11b.var3);
  Serial.print(F(","));
  Serial.print(sData11b.crc);
  Serial.print(F(","));
  Serial.print(rssi);
  Serial.println();
}
```

And here is some data that is being received, it clearly shows the same data packet twice. I have now tried for quite some time to identify the cause of the problem but am stuck.

Example output in Arduino serial monitor.
```
17:42:35.719 -> LoRa start
17:42:35.756 -> LoRa ready
17:42:35.756 -> READY_
17:43:04.346 ->  => 41,25022,650,3100,2260,180,-79
17:43:04.346 ->  => 41,25022,650,3100,2260,180,-79
17:43:30.916 ->  => 20,20000,2280,2900,0,92,-59
17:43:30.916 ->  => 20,20000,2280,2900,0,92,-59
17:43:32.955 ->  => 41,24001,480,3000,6050,112,-92
17:43:32.955 ->  => 41,24001,480,3000,6050,112,-92
17:44:05.657 ->  => 10,11018,-1700,0,0,129,-80
17:44:05.657 ->  => 10,11018,-1700,0,0,129,-80
17:44:36.586 ->  => 20,20000,2280,2900,0,92,-60
17:44:36.586 ->  => 20,20000,2280,2900,0,92,-60
17:45:22.588 ->  => 41,25022,660,3100,2290,32,-79
17:45:37.196 ->  => 41,24001,500,3000,6080,166,-92
17:45:37.233 ->  => 41,24001,500,3000,6080,166,-92
17:45:42.221 ->  => 20,20000,2280,2900,0,92,-61
17:45:42.221 ->  => 20,20000,2280,2900,0,92,-61
```

Does anyone have any suggestions about how to identify or fix this problem?
Thanks

IoTThinks, 2021-04-30:
You should try the default examples first.

gjt211, 2021-05-01:
@IoTThinks, yes I did thanks, they don't work either.

It turns out (after spending a lot of time studying how this library works and looking at many ofl the changes over time) that there is an issue with the MKNBR boards that I have worked out also seems to affect the MKRZero as well.
I have modified the library and have fixed the problem.

closed 2021-05-01 by gjt211

IoTThinks, 2021-05-01:
@gjt211 How did you fix the lirary for your board?
Thanks.

gjt211, 2021-05-01:
Hi @IoTThinks

My version of the library has a few small changes to suit our hardware, but basically the following is the library changes I made.
Yes it is a hack, but that's what I found and it works. Weird thing is, I have a product from about a year ago that has been working. It is possible that some MKRZero core changes may have introduced the issue I guess.

in LoRa.h I have added (for our hardware - we use SPI1, not SPI)
```
#elif defined(ARDUINO_SAMD_MKRZERO)
#define LORA_DEFAULT_SPI           SPI1
#define LORA_DEFAULT_SPI_FREQUENCY 200000
#define LORA_DEFAULT_SS_PIN        7
#define LORA_DEFAULT_RESET_PIN     6
#define LORA_DEFAULT_DIO0_PIN      10
```

And in two places in LoRa.cpp
int LoRaClass::endPacket(bool async) after
```
 // clear IRQ's
 writeRegister(REG_IRQ_FLAGS, IRQ_TX_DONE_MASK);
```
I added
```
    #ifdef ARDUINO_SAMD_MKRZERO
    writeRegister(REG_IRQ_FLAGS, IRQ_TX_DONE_MASK);
    #endif
```
and in int LoRaClass::parsePacket(int size) after
```
  // clear IRQ's
  writeRegister(REG_IRQ_FLAGS, irqFlags);
```
I added 
```
  #ifdef ARDUINO_SAMD_MKRZERO
  writeRegister(REG_IRQ_FLAGS, irqFlags);
  #endif
```

IoTThinks, 2021-05-01:
Thanks a lot. (y)