sandeepmistry/arduino-LoRa #468: lora communication

hoseinaghajari, 2021-05-17:
hi guys. I have confused these days by Lora Library. what I want to do is send and receive data with **SX1278 Module(https://cutt.ly/zbDX8K4)** and Arduino Uno. I can send data whenever I want and  the important thing for me is Low power consumption SO I can set my device in SLEEP_MODE after sending message. But I don't know how can I have low power consumption in receiving mode? 
see this: 
Rx..................>>
[image]

 in this mode the device is always ON. so how can I alert my device for receiving data? I mean me device always sleep until a receiving packet has arrive.

IoTThinks, 2021-05-22:
Use receivedCallback.
If DIO0 is high, wake the MCU up.

However, it will take 18mA when listening.

hoseinaghajari, 2021-05-23:
@IoTThinks  you mean that use **onReceive()** function? According this picture from library example. is it necessary to use **LoRa.receive();** after  **onReceive()** ?
[image]

**if yes; what about loop function? if I want to send message and waiting for answer, So it is necessary to call LoRa.receive(); after onReceive() In loop function? like this:
[image]

IoTThinks, 2021-05-23:
You can read duplex callback example.

Btw, there are two RX modes.

After sending a message, you must set LoRa.receive() if using RX continious (aka Receive callback).

If you use Single RX, call parsePacket as frequent as possible.

IoTThinks, 2021-05-23:
Note that you can only send or receive a packet at a time. Aka half duplex.

hoseinaghajari, 2021-05-23:
my problem is the sender cant receive well after sending message. just that.

IoTThinks, 2021-05-23:
Show the real photo of your board and antenna.

If already use the exact sample codes, then try to use an external power for LoRa. Likely insufficient power.

Arduino Nano?

hoseinaghajari, 2021-05-23:
**sender code:**

[image]

**receiver code.**

[image]

IoTThinks, 2021-05-23:
Read this again https://github.com/sandeepmistry/arduino-LoRa/blob/master/examples/LoRaDuplexCallback/LoRaDuplexCallback.ino

onReceive(LoRa.parsePacket()) is wrong to me.

hoseinaghajari, 2021-05-23:
**_

thanks to you. I test again and tell you the result.

_**

hoseinaghajari, 2021-05-23:
**how could I switch between tx mode and rx mode?
this is from datasheet. Does library do this?**
[image]

IoTThinks, 2021-05-24:
You can read the source code of this library.
Especially in BeginPacket (to send), ParsePacket(to receive in single RX) and LoRa.receive() (to receive in Continuous RX).

hoseinaghajari, 2021-05-24:
I tested last night and I had problem again. I think the library can't support this kind of communication. I send message to node2 and node2 send his answer ( with Answer(); ) to node1. but after 3 or 4 times it corrupt. 
**_Node1:_** 
[image]

**_Node2:_**
[image]

**_the result from node1:_**
[image]
or:
[image]

hoseinaghajari, 2021-05-24:
I have good power, I have good connection, but why? why it can send 6 or 7 time at first?

IoTThinks, 2021-05-24:
I believe you are a newbie in programming. No offence, btw.
You need to try and explore for yourself a while.
Use the exact examples at first.

BTW, the bellow codes may have problems.
[image]

and
[image]

and don't try to send inside an ISR.
[image]

hoseinaghajari, 2021-05-24:
it is same as library example, why you said its wrong usage?
**_Library example:_**
[image]

**_[image]_**

hoseinaghajari, 2021-05-24:
**Is it need sth to add while changing tx mode to rx mode?**

IoTThinks, 2021-05-25:
How many times the LoRa.receive() is called in your code and in the example code in 3 seconds?
LoRa.receive() is to set the LoRa into continuous RX mode.

You should find the datasheet of SX1276/1278 to read about the Single RX mode and Continuous mode.
Read the source code and try the **exact** example code.

> **_MY CODE_**
> **_[image]_**

hoseinaghajari, 2021-05-26:
hi guys. I fixed the problem and it work, as a reminder; node1 send hi to node2 and node2 answer it. it work now but take a look at my code please and tell me why ir stop working after 60 minutes? I Send 300 times at 1 hour. **after around 300 times sending, the Node2 cant receive message. why?**

**_Node1(Sender):_**

```
#include <SPI.h>
#include <LoRa.h>

unsigned char Hi[] = "Hi I'm Node1";
String inString = "";

void setup(){
  Serial.begin(9600);
  while (!Serial);
  Serial.println("LoRa gateway");
  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {

  if(runEvery(10000))
  Send(Hi);

   int packetSize = LoRa.parsePacket();
  if (packetSize){
  Receive();
  }
}

void  Send(String Byte){
 ​LoRa.beginPacket();
 ​LoRa.print(Byte);
 ​LoRa.endPacket();
 ​Serial.println(Byte);
 ​}

void Receive(){
    Serial.print("received:  ");
    String incoming = "";
  while (LoRa.available()) {
    incoming += (char)LoRa.read();
               }
Serial.println(incoming);             ​
        }  

  
  boolean runEvery(unsigned long interval){
  static unsigned long previousMillis = 0;
  unsigned long currentMillis = millis();
  if (currentMillis - previousMillis >= interval){
    previousMillis = currentMillis;
    return true;
  }
  return false;
} 

```

**_Node2(Receiver):_**

```
#include <SPI.h>
#include <LoRa.h>

void setup() {
  Serial.begin(9600);
  while (!Serial);
  Serial.println("LoRa Sender");
  if (!LoRa.begin(433E6)){
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
    int packetSize = LoRa.parsePacket();
  if(packetSize){
  Receive();
  Answer();
  }
}

void Send(String Byte){
  LoRa.beginPacket();
  LoRa.print(Byte);
  LoRa.endPacket();
  }

void Receive(){
  String incoming = "";
  while (LoRa.available()) {
    incoming += (char)LoRa.read();
  }
  Serial.print("Received: " + incoming);
  }

void Answer(){
    String sendANswer = "I got it";
     Serial.print("   Send: ");
               Serial.println(sendANswer);  
                Send(sendANswer);
}

```

**_after around 300 times sending, the Node2 cant receive message. why?_**

hoseinaghajari, 2021-05-26:
**_Dear @sandeepmistry please help me._**

IoTThinks, 2021-06-23:
May be related and solved issue: https://github.com/sandeepmistry/arduino-LoRa/issues/363

hoseinaghajari, 2021-07-12:
this problem was solved for me.

closed 2021-07-12 by hoseinaghajari