sandeepmistry/arduino-LoRa #471: Loss of communication after time

Jono72, 2021-05-25:
Hi sanddeepmistry,

Not sure if you can help, I have read issues that my problem may be related to, I am using library for a simple MQTT Lora node (that is going to be used in a private Lora network), I am sending data via MQTT to my node-red hub running on my pi via a dragino gateway, all seems to be fine and data is begin sent and decoded correctly, I also have a heartbeat that is sending back to the node every 30Secs, the idea being that if no heartbeat is detected an LED indicates loss of comms.

The Lora send is called in the main loop and just runs, every thing seems to be ok, but for some reason it just stops, sending and receiving, if I pull DIO2 down it sends, so I know its not my code that  is stopping it, rather than the Lora module just stops sending out on DIO2 pin.
Once I reset the whole thing resumes and runs from anything 2Hrs to 25Hrs, but always just seems to lockup.
I have tried external 3.3v power supply to ensure that the Lora module has enough power to send, I'm just using Lora module and Atmel Mega board for prototyping atm as I design and build my own boards once everything is working.

My code is rather large to put up here as it has multiple files, however I do not use any blocking code or delays and the actual program still runs (just no send or receive): below is a snippet from the Lora send and receive part, I cant see why the Lora module just stops. 

#include <LoRa.h>
#include "Global.h"
#define RE(signal, state) (state=(state<<1)|(signal&1)&3)==1

float tem,hum;
char tem_1[8]={"\0"},tem_2[8]={"\0"},tem_3[8]={"\0"},tem_4[8]={"\0"};   
uint8_t datasend1[40];
uint8_t datasend2[36];
unsigned int count = 1; 
unsigned long new_time,old_time=0;
char command [64];
char topicIn  [64];
static int RiseDI;

//Called once in setup
void loraSetup()
{
	
     
      Serial.println(F("Start MQTT Example"));
      if (!LoRa.begin(915000000))   //868000000 is frequency
      { 
          Serial.println("Starting LoRa failed!");
          Blink2(ALM, 500, 500); 
	  }
      // Setup Spreading Factor (6 ~ 12)
      LoRa.setSpreadingFactor(7);
      
      // Setup BandWidth, option: 7800,10400,15600,20800,31250,41700,62500,125000,250000,500000
      //Lower BandWidth for longer distance.
      LoRa.setSignalBandwidth(125000);
      
      // Setup Coding Rate:5(4/5),6(4/6),7(4/7),8(4/8) 
      LoRa.setCodingRate4(5);
      LoRa.setSyncWord(0x34); 
	 
      Serial.println("LoRa init succeeded.");
      
      LoRa.onReceive(onReceive);   
      		
	
}

void loraformat()
{
 
	//Convert temps to strings
	dtostrf(AI1,0,1,tem_1);
	dtostrf(AI2,0,1,tem_2);
	dtostrf(AI3,0,1,tem_3);
	dtostrf(AI4,0,1,tem_4);
	
	
	
	//Format datasend packet
	sprintf(datasend1,
		"<%d>"
		"%s,%s,%s,%s," //Temperatures
		"%d,%d,%d,%d,%d,%d,%d,%d", //DigitalIO
		NodeID,
		tem_1, tem_2, tem_3, tem_4,
		DigitalIO[0], DigitalIO[1], DigitalIO[2], DigitalIO[3], DigitalIO[4], DigitalIO[5], DigitalIO[6], DigitalIO[7]
	);
	
    
	
	
}
//Lora send called in the main loop
void loraSenddata()
{
	
	static bool state;
	
	//The following only executes send evry 8 secs
	static uint64_t timer = currentMillis;
	
	
	//Heart beat to stop sending if Heartbeat time out expires commented out for testing purposes
	//if (heart)
	//{		
		if ((currentMillis - timer) >= 100)
		{
			state = LOW;
		}
		if ((currentMillis - timer) >=100 + 8000){
			state = HIGH;
			timer = currentMillis;
		}	
		//digitalWrite (CON, state);	
	
		if (state)
		{
			LoRa.beginPacket();
			LoRa.print((char *)datasend1);	 
			LoRa.endPacket();
			Serial.println("Packet Sent");
			// print RSSI of packet
			Serial.print("' with RSSI ");
			Serial.println(LoRa.packetRssi());	
			RSSI = LoRa.packetRssi();
		}
	
	//}
	LoRa.receive();
}

void onReceive(int packetSize) {

  char msg [32];
 
	
  Serial.print("Received packet : ");

  // read packet
  for (int i = 0; i < packetSize && i < 31; i++)
  {
	 msg[i] = LoRa.read();
	 msg[i + 1] = '\0';
  }
  
  
  
  Serial.println(msg);
  
  //Relay 1 Enable Command
  char DOstring[32];
  sprintf(DOstring,"%d/%s" ,NodeID, "DO1_ON");

  if (strcmp(msg, DOstring) == 0)
  {
    DigitalIO [4] = HIGH;
    Serial.println ("Digital Output 1 ON");
	
  }  
  sprintf(DOstring,"%d/%s" ,NodeID, "DO1_OFF");
  
  if (strcmp(msg, DOstring) == 0)
  {
    DigitalIO [4] = LOW; 
    Serial.println ("Digital Output 1 OFF");
  } 
    //Relay 3 Pulse On Command
	
	sprintf(DOstring,"%d/%s" ,NodeID, "DO3_PULSE"); 
  
  if (strcmp(msg, DOstring) == 0)
  {
    (DigitalIO [6]) = true;

    Serial.println ("Digital Output 3 ON");	
	
  }
  
// Heartbeat Rcv String	
  sprintf(DOstring,"%d/%s" ,NodeID, "heart"); 
  

  if (strcmp(msg, DOstring) ==0)
  {
	 Serial.println ("Heart Rcvd");
	 heart = 1;
	 Serial.println(heart);
	 
  }
  
}

IoTThinks, 2021-05-25:
I'm not Sandeep =))
1. I don't think DIO2 is related. By right, it should be for the chip to signal out the event.
However, I believe this library code doesn't touch any setting for DIO2

[image]

My advice is to use the exact example codes to run for a while to see if they stop working first.

2. After step 1, if you search in this github, there are a few people complain of similar issue? https://github.com/sandeepmistry/arduino-LoRa/issues/378

Jono72, 2021-05-25:
Hi Ok  thanks (sorry bout the name thing :)

Sorry I wasn't clear, When I said DIO2 I ment the input to the Pin 2 on the Mega, I believe that comes from DIO0 on the RFM module (RxDone), If I pull that pin Low when its locked up, the Mega sends once, when I reset all is good. the shield used the RF96 chip, I plan to use the HOPE RF module in my project (100mw), is the library compliant with that module?

 But thanks I will try your advice and see what happens.

Regards

John

IoTThinks, 2021-05-25:
HopeRF RFM95/96... will work.

Yes, try the examples (sender and receiveCallback) first to see if the issue happens.

Then can try to send async by endPacket(true) to ensure no blocking. 
Then view similar issues here.

I mainly use esp32 boards. And so far, they can send and receive for months.

Ok try and see.

Jono72, 2021-05-25:
Hi IoTThinks

Will do thanks for the advice :)
- [ ] 

- [ ]

IoTThinks, 2021-06-23:
Related and solved issue: https://github.com/sandeepmistry/arduino-LoRa/issues/363

IoTThinks, 2021-07-04:
For reference: https://github.com/sandeepmistry/arduino-LoRa/issues/307