sandeepmistry/arduino-LoRa #474: Starting LoRa failed! ESP8266 & RFM95W Version = 0

RubiRubsn, 2021-05-28:
Hi Guys,
I´ve tried to play a bit around with a RFM95W and a ESP8266. I wired the RFM to the ESP like that: SS->D8, SCK->D5, Mois->D7, Miso->D6, Dio0->D1, Reset->D2
I`ve used this example:
`
#include <Arduino.h>
#include <SPI.h>
#include <LoRa.h>

int counter = 0;

void setup()
{
  sei();
  Serial.begin(9600);
  while (!Serial)
    ;

  LoRa.setPins(D8, D2, D1);
  Serial.println("LoRa Sender");

  long FREQ = 915000000;
  if (!LoRa.begin(FREQ))
  {
    Serial.println("Starting LoRa failed!");
    while (1)
      yield();
    ;
  }
}

void loop()
{
  Serial.print("Sending packet: ");
  Serial.println(counter);

  // send packet
  LoRa.beginPacket();
  LoRa.print("hello ");
  LoRa.print(counter);
  LoRa.endPacket();

  counter++;

  delay(5000);
}
` 
I'am using PlatformIO.

I get a "Starting LoRa failed!" in the Serial monitor and after adding 
`  
uint8_t version = readRegister(REG_VERSION);
  Serial.print("version: ");
  Serial.println(version);
`
im getting "Version: 0"; 
Dumping my Register shows, the complete register is 0.

Maybe i have a problem with a connection but i double and Tribble checkt everything.

Thanks for your help^^

RubiRubsn, 2021-05-30:
found my Problem, had a cold soldering point. (on two modules, it was hard to find) thx for your help. :D

closed 2021-05-30 by RubiRubsn

ghebra, 2021-06-20:
please help me , how you are resolve this issue