sandeepmistry/arduino-LoRa #475: Baud Rate for LoRa SX1276?

CKtrac, 2021-06-01:
Hi,
I am using the K22F as a Microcontroller and wanted to know if there is a recommended baud rate for communicating with the LoRa module via SPI? I have the SPI-Frequency currently at 8 MHz but still don't get the right responses from the LoRa. From the Arduino Code I would have assumed that the baud rate is the same as the SPI-Frequency but that is not working.
Thanks in advance!

IoTThinks, 2021-06-11:
Baud rate? Are you using an UART LoRa module instead?

CKtrac, 2021-06-14:
Hi, am using the SX1276 mbed shield but I found the problem last week. Baudrate of 8Mhz is fine the problem was in the clock frequency of the Microcontroller and the configurations for the full duplex protocoll that I had wrong. So you can close this issue.

CKtrac, 2021-06-14:
I have another question though and am unsere whether or not I should open another issue. Now that the LoRa communication is working I want to build a LoRa WAN Stack. In the FAQ it is mentioned that this library exposes the LoRa radio directly and that the LoRaWAN library builds on top of it. Now there are several LoRaWAN libraries out there. Do you know of one that is compatible with this LoRa library?

IoTThinks, 2021-06-14:
For LoRaWAN, use this https://github.com/mcci-catena/arduino-lmic
LoRaWAN is a protocol stack not just LoRa sending/receiving.

closed 2021-06-14 by IoTThinks

CKtrac, 2021-06-17:
Sry to bother again, but what does this mean for this Repo? I always read that LoRaWAN is a protocol stack "on top of" the LoRa physical layer. But does this mean exactly? How do I build a LoRaWAN on top of LoRa?
Do I need to use both Repos this here and https://github.com/mcci-catena/arduino-lmic and merge them somehow or do I only need the LoRaWAN Repo and it handles the LoRa communication as well?

IoTThinks, 2021-06-18:
> Sry to bother again, but what does this mean for this Repo? I always read that LoRaWAN is a protocol stack "on top of" the LoRa physical layer. But does this mean exactly? How do I build a LoRaWAN [...]

Please raise the question in another topic.
The question is not really relevant with the original question.