sandeepmistry/arduino-LoRa #480: LoRa vs LoRaWAN

CKtrac, 2021-06-20:
In the FAQ it is mentioned that this library exposes the LoRa radio directly and that the LoRaWAN library builds on top of it. I know that LoRaWAN is a protocol stack not just LoRa sending/receiving. But what does this mean for this Repo? I always read that LoRaWAN is a protocol stack "on top of" the LoRa physical layer. But does this mean exactly? How do I build a LoRaWAN on top of LoRa?
Do I need to use both Repos this here and https://github.com/mcci-catena/arduino-lmic and merge them somehow or do I only need the LoRaWAN Repo and it handles the LoRa communication as well?

IoTThinks, 2021-06-21:
You can use LMIC alone for LoRaWAN. LoRa implementation is already inside.

If you insist to use this library for LoRaWAN, then go ahead and put in your own code to "reinvent the wheel".

CKtrac, 2021-06-21:
OK, thanks. Then I have a last question before I decide which repo to use:
I want some kind of "authentication" for the Communication. So my goal is to have e.g. 2 LoRa Senders (A and B) and several LoRa recievers (let's say 2 for each Sender, so A1 A2 B1 and B2). I want to make sure that the Recievers do only recieve from the Sender they belong to and not from the other Sender. So that A1 and A2 only reciever or read from Sender A and not Sender B. To do this I will probably have to implement my own code that handles this right? I have seen some features in the LoRa like the implicit header mode, but I guess this is not what I need here, right? Or is there anything in this repo I can use for this that I have not recognized yet?

IoTThinks, 2021-06-21:
LoRa is at Physical layer.
Except sending and receiving LoRa messages, you need to do everything on your own.

Implicit/explicit header is not related to authentication.
It only contains packet length...

CKtrac, 2021-06-21:
ok, thanks a lot!

closed 2021-06-29 by IoTThinks