sandeepmistry/arduino-LoRa #482: Not able to send larger payloads (>200byte)

mcr-ksh, 2021-06-25:
Hi,

i've been testing and sending data from T-Beam V1.1 without a problem if the data size is below approx. 200 bytes. If I exceed somewhat around the 200 bytes the packet is not transmitted (or received) on my receiver.  Smaller packages are received.

Sender: T-Beam V1.1 
Receiver: ATmega328p + SX1279 

Anyone else experiencing this kind of issue?

mcr-ksh, 2021-06-25:
1. Does the "why" really matter? -> it's a requirement
2. Crc is already enabled. I'm afraid I don't understand the part about the sending interval as I write to the SPI register the bytes to be send and then trigger the register (function endPacket) to send the package.

mcr-ksh, 2021-09-13:
That is very helpful for LoraWAN. Here im talking about direct lora access to the chip. So the limits should not apply if not enforced by the chip

Kongduino, 2021-09-14:
The limits apply in the sense that while there may be some regulatory restrictions originally (Fair Use yada yada), in the (H) range (868+) at least, dwell time limitations (400 ms max) impose a physical limit on the amount you can actually transmit. A chip that ignores that will not get FCC certification.

In older chips, SX127x, in (L) frequencies (433 etc) there are supposedly no dwell time restrictions, but somehow sending larger packets doesn't always work. Also, remember that the 255 limit is the WHOLE budget, including preamble, header, header CRC, and possibly payload CRC. So altogether, between the overhead, and the physical restrictions due to the SF/BW combination, your packets could be too long to send, and hang the chip. My advice would be to:

(a) reduce the size, say 150, then add a few bytes each time and see when it break.
(b) use different combinations of SF/BW (reduce SF, raise BW).

TA2SVS, 2024-01-26:
hello @mcr-ksh I am sorry for disturbing you. I'm having the same problem and I couldn't find the crc related code anywhere. Is there any way you can share it? . I don't know how it's done. Thank you.

mcr-ksh, 2024-01-27:
> hello @mcr-ksh I am sorry for disturbing you. I'm having the same problem and I couldn't find the crc related code anywhere. Is there any way you can share it? . I don't know how it's done. Thank you.

Too long ago. I have no idea anymore. sorry.