sandeepmistry/arduino-LoRa #483: Resetting radio periodically

beowulff, 2021-06-27:
I would like to reset the LoRa module periodically (maybe every 15 minutes), to try to prevent lock-ups.
Is there any issue with calling **LoRa.begin** over and over again? Will that cause a memory leak?

IoTThinks, 2021-06-29:
May be something is wrong with Arduino SPI?
I never have lock-up with esp32.

BTW, I think no memory leak issue to reset LoRa frequently.
I used to end and begin LoRa periodically.
Later I even turn off the power of the whole LoRa module.

beowulff, 2021-06-29:
I'm not sure there is any problem, but this is for a commercial product, and it needs to be reliable.
If I can reset the radio on a regular basis, that should guard against some forms of lockup.
I'm testing it right now, calling my init_LoaRa function (which calls LoRa.begin) every 1 minute. It's been doing that for a few hours, and seems to be working fine.

If I had a spare port pin, I would control the power to the radio from the ESP32, but I'm fresh out of pins...

IoTThinks, 2021-06-29:
Is your board STM32 or Arduino-based?
I believe there is no memory leaking issue.
However, resetting so many times may have wear-and-tear problem with electronics?

beowulff, 2021-06-29:
It's ESP32.
I wouldn't think there is any wear issue - I'm not writing to Flash in any of the init routines.
I might change the reset scheme to one where the radio is only reset if I don't get a "heartbeat" communication (which happens every 10 minutes). I would reset the entire system, but that might cause issues with devices being controlled by the ESP32.

closed 2021-07-04 by IoTThinks