sandeepmistry/arduino-LoRa #484: Range problems

CKtrac, 2021-06-28:
Hello, I am trying to send between 2 LoRa modules connected with an Arduino Mega 2560 each. I tinkered around with various settings and currently I am using:
-setTxPower(20)
-LNA gain 0x01
-auto AGC to 0x00 (so that manual LNA gain is used)
-signalbandwidth(125E3)
-spreadingfactor(6) and I also tried spreadingfactor(12)
- my payload size is currently 252 bytes (I also tried 25 bytes at one point and increased the range to 15m but that's it)

my Problem is that at the moment the sending and recieving only works within 3m or so and if I move further away I do not recieve anything. From an older post I found (https://github.com/sandeepmistry/arduino-LoRa/issues/432) it could be a hardware problem which is why I attached my setup to this post.
The recieved RSSI value is always ~ -109
The recieved SNR is at 1.75 at close distance and decreases to up to -20 when far aways (setup with 25 bytes payload and Spreadingfactor 12 and distance ~10m)

How can I increase the range? What is the problem with my setup?
[image]
[image]

beowulff, 2021-06-28:
You have a bad antenna, or antenna is not connected.
Remove you antennas, and see if the range gets worse than it already is.
I can easily get 2.5 miles with the correct stubby antenna.

beowulff, 2021-06-28:
Also, make sure you are setting the module to the 915MHz band.
If it's set to 433MHz, it doesn't have an antenna connected...

CKtrac, 2021-06-29:
The module is set to 868Mhz, I forgot to mention that. Is the antenna fit for 868Mhz? I will try it out without the antenna today.

CKtrac, 2021-06-29:
I just tried it out without the antenna and the situation is not improving. If I remove the antenna from the sender it does not make a difference at all. If I remove the antenna from the reciever, it is not even recieven when the modules are lying next to each other. This situation is the same for 868Mhz as well as 915Mhz. I now used the spreading factor 7

CKtrac, 2021-06-29:
Ok, I found the solution. Turns out I was using the wrong antenna. Now I changed it to antennas fit for 869Mhz and it works so much better now. Thanks for the hint!

CKtrac, 2021-07-05:
Ok, I celebrated too soon. Turns out that even though the situation improved from the initial 3m I stop recieving after ~100m. I even went back to the default settings to make sure I did not ruin anything:

-Frequency at 868E6
-setTxPower(27)
-LNA gain 0x03
-auto AGC to 0x04
-signalbandwidth(125E3)
-spreadingfactor(7)

I am using the correct Antenna for the frequency I use (see picture). I decreased the payload to a simple "hello" and was able to get 10m more but that is way too less for me. I read in various posts, that people are able to send across several km even within a city, yet I am unable to achieve the same performance. What is wrong with my setup?

[image]

CKtrac, 2021-07-05:
To add to this. My RSSI ist most of the time around -110 the SNR starts usually at ~4 and then drops to ~ -7

IoTThinks, 2021-07-05:
Try to use external 3.3v power for LoRA

IoTThinks, 2021-07-05:
Your SNR and RSSI implies either wrong antenna or not enough power.

To check if any jumper for the antenna

CKtrac, 2021-07-05:
Ok, I have checked the following things now:
- I used a level shifter and checked with an oscilloscop that I am using 3.3v to power the LoRa
- I set LNA Gain to 0x00 and AGC to 0x04 so that I use auto AGC
- I am using the 869Mhz Grey Rings antenna that was inside the Packet for the transmission
Still nothing has improved. I am unable to recieve any transmission after ~100m. Is there anything else I could check? Thanks for all the help so far!

IoTThinks, 2021-07-05:
Check if you mix up between the SMA and RP-SMA antenna?

One of the connector should be male and the other is female.

beowulff, 2021-07-05:
If I were you, I would make my own antenna, using the correct length wire.
This is what I did when I was first trying to get the longest possible range.
A photograph of the antenna I made is here: https://github.com/sandeepmistry/arduino-LoRa/issues/336

Unless you have access to a VNA or Spectrum Analyzer, there's no way to know if the antennas are any good, unless you make your own.

CKtrac, 2021-07-05:
I checked the male and female connectors so that they fit. The Antennas are both male and on the LoRa module they are female (see picture). Is this what you meant?

I might be able to get access to a VNA tomorrow or at the end of the week. Assuming that the output of the antenna is bad, what could I do then? Is there anything in the code or from a setting-perspective that I could change? Or is the only solution to get another antenna? I got the antenna from Semtech fitting for the LoRa mbed shield and it would be wondering if they don't fit, or does this happen often?
[image]

CKtrac, 2021-07-05:
One thing to mention is that I am in Europe and that I am using the SX1276 mbedh shield "MB1MAS". I just stumbled upon this link: https://os.mbed.com/users/dudmuck/code/chat_sx127x/wiki/Homepage were it is written that the European version does not have this PA_BOOST pin option to devliever more than 14 dBm. Could this be my problem? If so, should I not be able to reach further with up to 14 dbm?
You were mentioning something with jumpers and the antenna. What was that?

IoTThinks, 2021-07-06:
Some boards have special jumper.
If it doesn't support PA BOOST then try LFO instead

LoRa.setTxPower(txPower, PA_OUTPUT_RFO_PIN);
https://github.com/sandeepmistry/arduino-LoRa/blob/master/API.md#tx-power

CKtrac, 2021-07-06:
Thx for the hint: LoRa.setTxPower(txPower, PA_OUTPUT_RFO_PIN). I am sorry that I missed that, now I could improve again the range. I was able to reach up to 380m with 14dbm (maximum without PA BOOST). This also means, that I will for now continue to use the antenna that was send with the kit and save the option of building my own antenna for later (if everything else is not working T_T)

My RSSI and SNR are still bad (-130 and -12.50) but I assume that this is because I am trying this out in the city. However, I have access to two rooftops that are 800m apart and was not able to recieve at this distance.
I am using now the bandwidth of 31.25E3 and the Spreadingfactor of 12 for the longest possible range. From what I read in the datasheet, a smaller bandwidth and higher SF increase the range. Are this the right settings, or can I still change something to increase the range? What would be the optimung settings for longest range?

IoTThinks, 2021-07-07:
No need to stay too far. Sometimes, it takes a lot of effort to climb up the roof top.

If within 3-5 meters with no obstacle and you can not get RSSI around -60 to -80 / SNR around +10, then it is likely antenna and antenna related issues.
How much RSSI / SNR do you get in 3-5 meters with no obstacle?

CKtrac, 2021-07-12:
at 0.5m I have RSSI = -65 and SNR=10
at 3m I have RSSI = -87.5 and SNR=varying from 9 till 12
at 5m I have RSSI = -101 and SNR= 8

IoTThinks, 2021-07-12:
If no obstacle in 3m, and the RSSI is around -80, I think the antenna is badly tuned.

It should be around -60.
Can try to do your own antenna.

frequently-incorrect, 2023-11-09:
I had the same issue with the SX1276MB1MAS until I figured out that the TXRX_SW is by default not connected to the RXTX/RFMOD pin on the transceiver. This means that the actual antenna never gets connected to the rest of the circuit when transmitting which results in the issues you and I have experienced.

You can fix this in two ways:

1. Add a blob of solder at the label R15 in the MB1MAS schematic, connecting the two pins.
2. Send a high signal to the MB1MAS breakout pin "RxTx" (Arduino pin 18) whenever you want to transmit and a low signal whenever you want to receive.

I used option nr. 1 and my RSSI increased from -90 dBm at ~1m to -15 dBm at the same distance.

I realize this is probably too late for helping you @CKtrac but I'm leaving this solution for anyone else arriving here after struggling like me. Good luck!