sandeepmistry/arduino-LoRa #485: relation between Distance and (SF & BW & CR)

hoseinaghajari, 2021-06-30:
Hi there, what is exactly relation between **Distance**(between 2 module) and **Values of SF & BW & CR**?

IoTThinks, 2021-06-30:
There is no such mathematical equation for wireless world.

morganrallen, 2021-06-30:
Sure there are calculations that can be done. In the RF world a 'Link Budget' calculation is a standard practice to get an idea of Link-to-Link performance (including theoretical distance).
SemTech even provides the 'LoRa Modem Calculator Tool' to give you a lot of detailed information. This article give a pretty good overview of understanding these kind of calculations but always understand that there will be a lot of variance between different setups. https://medium.com/home-wireless/testing-lora-radios-with-the-limesdr-mini-part-2-37fa481217ff

IoTThinks, 2021-07-04:
There is calculation.
But not an exact caculation as wireless is not a perfect environment.

The calculation is only for a perfect no-obstacle environment.

IoTThinks, 2021-07-04:
Can view more from this? https://youtu.be/jMFa3AiDbcI

Kongduino, 2021-07-04:
As mentioned, there isn't an exact way to calculate the impact. However, as a rule of thumb, the higher the SF, and the lower the BW, and the slower, and longer range, the transmission will be. And vice versa.