sandeepmistry/arduino-LoRa #486: examples are very poor / do NOT demonstrate how to store received data into variables

StefanL38, 2021-07-03:
I have looked into the examples.
The examples do somehow andvanced stuff.
But **none** of the examples does a demontsration on how to receive two values 
how to extract them from the received message and store them into variables.

This is very beginner-**UN**-friendly

IoTThinks, 2021-07-03:
Then try yourself harder.

morganrallen, 2021-07-03:
@StefanL38 this library is very low-level and the example are meant show only the most basic functionality.
Some more practical examples ***would*** be super cool but I personally don't have the capacity these days.

> Then try yourself harder.

that's not helpful for anyone. seriously, check your attitude, you're here to help folks right?

beowulff, 2021-07-04:
Once you have parsed the LoRa packet, you can just call LoRa.read to read bytes into your variables.
Just as you would do with a file.

IoTThinks, 2021-07-04:
@morganrallen It is OSCP's motto: Try harder.

Sometimes, we need to try harder instead of simply blaming all other stuffs too difficult to do.

morganrallen, 2021-07-06:
And if this was OSCP that would mean something. But as it's not, just chill out on the harsh criticisms, they don't actually help anyone and you don't *have* to answer anyhow.

closed 2021-07-06 by morganrallen