sandeepmistry/arduino-LoRa #487: switch between rx and tx

hoseinaghajari, 2021-07-04:
hi dear friends:
**my project is sending a message to receiver and receiver sending answer to sender.**
I use 2 arduino uno and 2 RA01-sx1278.
my connections are correct too.
I have problem with Lora receiver; after some hour I cant receive anything. but sender works well.

this is my sender code:
```
#include <SPI.h>
#include <LoRa.h>
#include <PollingTimer.h>

PollingTimer timer;
long Interval=15000;
int i = 0;
String Hiall = "Hiall,02THL";
String message="";

void setup(){
  Serial.begin(9600);
  while (!Serial);
  Serial.println("LoRa gateway");
  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
    timer.start();
}

void loop() {
 if (timer.msec()>Interval){
      timer.start();
      LoRa.beginPacket();                                     //send message
      LoRa.print(Hiall);
      LoRa.endPacket(true);        
      delay(1000);
    }
 int packetSize = LoRa.parsePacket();
  if (packetSize) {
     message = "";
    while (LoRa.available()) {                                   //receive answer
      message += (char)LoRa.read();
    }
          Serial.print("     Receive: ");
          Serial.println(message);
 }
}
```
**#include <PollingTimer.h>** is just a timer library.

this is my receiver code:
```
#include <SPI.h>
#include <LoRa.h>
#include <PollingTimer.h>

PollingTimer timer;
int i = 0;
String message="";

void setup() {
  Serial.begin(9600);
  while (!Serial);
  Serial.println("LoRa Node");
  if (!LoRa.begin(433E6)){
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() { 
    int packetSize = LoRa.parsePacket();
  if (packetSize) {
      message = "";
    while (LoRa.available()) {
      message += (char)LoRa.read();                 //receive message
    }
      Serial.print("Receive:");
      Serial.println(message);
      delay(3000); 
	  
     LoRa.beginPacket();
     LoRa.print("HiN1,");                          //send answer
     LoRa.endPacket(true);
      delay(100);
  }
}
```

after some hours receiver can't receive anything. why? how can I have a half-duplex system? I tried half-duplex example but it didn't work correctly.

IoTThinks, 2021-07-04:
You can try this instead https://github.com/sandeepmistry/arduino-LoRa/blob/master/examples/LoRaDuplexCallback/LoRaDuplexCallback.ino

"delay" is the enemy for parsePacket. Your node may be stuck in the delay for 3s and can hear anything during the delay.

```
LoRa.endPacket(true);
delay(100);
```
=> Try to use LoRa.endPacket() instead.

hoseinaghajari, 2021-07-05:
dear @IoTThinks ; I tested it with **LoRaDuplexCallback.ino** how much exactly **power** needed for RA01-sx1278? I connect it to 3.3v of UNO but after some hours the receiver cant receive. I guess it's because of power.

IoTThinks, 2021-07-05:
Sometimes, the LoRa chips will need up to 150mA (or around that) for TX (sending).
Some Uno and Mega can not reliable take the load.

There is two things you can try
1. Try to power your LoRa via external 3.3v. Especially at the sender. This is the most common issue. 
2. Can try this fix to see if it works. https://github.com/sandeepmistry/arduino-LoRa/issues/307#issuecomment-586899795

hoseinaghajari, 2021-07-12:
thank you @IoTThinks . It solved. I had power issue and code.

IoTThinks, 2021-07-12:
@hoseinaghajari What did you do to fix your problem?

hoseinaghajari, 2021-07-12:
I'd been using just parsepacket() and some daley() functions in it but now I use onReceive() and LoRa.receive() functions.

IoTThinks, 2021-07-12:
Great

closed 2021-07-12 by hoseinaghajari