sandeepmistry/arduino-LoRa #488: Enhancement request: interrupts on FALLING edge

Addictek, 2021-07-05:
I've designed a new circuit using ESP8266 and SX1278.

IO0 output of SX1278  is connected to GPIO2 input pin of ESP8266 through an inverter gate (NOT). This is required by ESP8266 since the GPIO2 pin must be pulled up on startup and the IO0 pin of SX1278 is low on startup.

Due to the inverted connection between IO0 and GPIO2, to use the onReceive() callback, I would need to attach the interrupt on the FALLING edge of GPIO2.

The current implementation only supports interrupts on the RISING edge. 

The request is to make the interrupt type configurable keeping the default value to RISING.

IoTThinks, 2021-07-06:
Generally, you should avoid using GPIO0 and 2 of esp32/esp8266.
You should fix your board design instead.

I believe the SX127x will trigger a HIGH signal on DIOx of the chip.
When the event is processed, then the chip will set DIOx back to LOW signal.
I wonder if your suggestion will ever work.

Addictek, 2021-07-06:
I used GPIO0/2 because I ran out of i/o pins. You are right, SX127x chip
triggers HIGH signal but in my case this is going through a mosfet gate
that inverts the signal before reaching the GPIO2 pin. This way, ESP chip
starts up regularly, but I can still use GPIO2 to handle the interrupt.
Note that the mosfet gate circuit is also able to keep the output high
regardless the input signal on start up for the time required by esp8266 to
boot.
I have modified the library for testing purposes and it is working fine.

Addictek, 2021-07-07:
Just to add that my change was something like:

#ifndef LORA_ISR_MODE
#define   LORA_ISR_MODE   RISING
#endif

and then using LORA_ISR_MODE in place of the hardcoded value in the library code. This will have no impact on projects currently using the library.