sandeepmistry/arduino-LoRa #489: RESET pin -1

Samuel-ZDM, 2021-07-07:
Hello!

I'm using an Arduino Nano with the RFM95 module. I have a problem when I set the RESET pin to -1. There is an error in the connection.

If I put the RESET on a pin, on pin 6 in my case, the error doesn't appear anymore, but it partially works, because it takes a long time to receive the packets.

NSS: 10
DIO0: 2
DIO1: 3
RESET: -1

Thank you very much in advance.

morganrallen, 2021-07-07:
What's the error? It should be able to handle -1 as the pin

Samuel-ZDM, 2021-09-16:
Hi! Sorry for the delay.
The error occurs when I set the RESET pin to -1, in the Serial Monitor it is shown that the LoRa connections are not correct. 

`const int csPin = 10;
const int resetPin = -1;
const int irqPin = 3;`

When I set the RESET pin to 6, the error goes away, but my device doesn't work. The device sends the first message, but stops, it seems the device locks up. I soldered pin 6 of the Atmega328 to the RESET of the LoRa module.