sandeepmistry/arduino-LoRa #493: LoRa.print() stopps working after first time if LoRa.sleep() is activated

chmasi, 2021-07-09:
Dear All 

This is my Code to send Lora Data
`
void loop() {

    //updateData();
    sendLoraData();

    for (int i = 0; i < 2; i++)
    {
      LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF);
    }

}

void sendLoraData()
{

    LoRa.idle();
    DBG_OUTPUT_PORT.print("Sending packet: ");
    DBG_OUTPUT_PORT.println(counter);
    //Send LoRa packet to receiver
    LoRa.beginPacket();

    LoRa.print('\0');   // Wrap update string
    LoRa.print("<");
    LoRa.print(myData.sensornr);LoRa.print(",");
    LoRa.print(myData.temp);LoRa.print(",");
    LoRa.print(myData.humi);LoRa.print(",");
    LoRa.print(myData.pres);LoRa.print(",");
    LoRa.print(myData.alti);LoRa.print(",");
    LoRa.print(myData.vraw);LoRa.print(",");
    LoRa.print(myData.airq);LoRa.print(",");
    LoRa.print(myData.vcc);LoRa.print(",");
    LoRa.print(myData.counter);
    LoRa.print(">");

    LoRa.endPacket();

    //LoRa.sleep();
 
    DBG_OUTPUT_PORT.println("data send finished");
    DBG_OUTPUT_PORT.flush();
    delay(100);
}`

This works quite fine,as soon I activate LoRa.sleep(), only the first time is working, after the second time it stops working and I have to manually reset the Arduino Pro Mini 3.3V 8 MHz

any suggestion how I can solve this issue?

IoTThinks, 2021-07-10:
I believe the problem is how you make your MCU sleeps.
LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF);

chmasi, 2021-07-14:
> I believe the problem is how you make your MCU sleeps and wakes up.
> LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF);

I changed the code to delay(xxxx) instead of the LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF) but result is the same after I put LoRa to sleep

naveen-tach, 2024-03-16:
hi team, I looking for lora. print, "print" definition, please suggest me where I can find in this lib,