sandeepmistry/arduino-LoRa #494: How can we wake up a node for receive packet?

hoseinaghajari, 2021-07-12:
hi dear friends:
my project is sending a message to receiver and receiver sends answer(acknowledge) to sender.
I use 2 arduino uno and 2 RA01-sx1278.
**what I want to have is _lowest power consumption_** so for doing that I tried to set sender to sleep mode and wake it up just before sending a message. _Notice that I know when I want to send a message just in sender side_ so I can manage sleeping and waking function in sender side. but what I should do for receiver? I can set it to sleeping mode after receive_and_send_acknowledge **but my problem is that when I should wake receiver up? The receiver don't know when I want to send a packet to it.**

IoTThinks, 2021-07-13:
You need to do it on your own if you want to use raw LoRa.
You can take a look at LoRaWAN A, B and C to get / copy the idea.

If you don't have any server to cache the sending packets, then likely you need to sync the time between the two nodes.

hoseinaghajari, 2021-07-18:
A or B or C LoRaWAN®-based network is made up of end devices, gateways, a network server, and application servers. But I only have end device and gateway. 
End device send data to gateway. Do gateway always have to stay awake?
gateway send data to End device. Do End device always have to stay awake?
certainly I don't want them to be always awake.

hoseinaghajari, 2021-07-18:
**how much current consumption of continuous receive mode?**

hoseinaghajari, 2021-07-21:
any help?

IoTThinks, 2021-07-21:
Google: LoRaWAN Class A, B C. 
Google SX1278 data sheet. Search for Power consumption.

Choose the first link.
BTW, your question is not really relevant to the library.

closed 2021-07-21 by hoseinaghajari