sandeepmistry/arduino-LoRa #495: Problem with setting Tx power

Nehuix, 2021-07-13:
Hi guys! I am currently using two sx1278 with the spiral antenna that usually comes with it. I am using the basic _LoRaReceiver_ and _LoRaSender_ examples and my problem is that even though they are side by side, I get an RSSI between -50 and -60. I have tried changing the output power with _setTxPower()_ with values between 1 to 14 and there seems to make no difference on the RSSI values.
I wonder if I am doing something wrong in the code or is it because of the antenna.
Thanks in advance.

IoTThinks, 2021-07-13:
Can you show your real photo of kit?
What is your LoRa chip? ra01/02/RFM?
What is your LoRa antenna?
What is the signal frequency?

Normally, with the best antenna for ra01/02 at 433Mhz, I can only get around -30 to -40 RSSI max.
For RFM95 at 915Mhz, I can get up to -6 RSSI.

However, it doesnt make any real difference to me at around 3-5m and above.

Nehuix, 2021-08-03:
Sorry I just replied. I've been travelling and couldn't respond any sooner. I don't currently have the circuit with me but this is the module I am using (I welded it to a propper pcb).
[image]

I'm also working at 433MHz and the antenna is the same as the picture.
I also bought a 16dbi antenna and the UFL joing for the module, but still haven't recieved it.
[image]

Hope this info is usefull.
Thanks in advance.