sandeepmistry/arduino-LoRa #506: Starting LoRa failed

surajgawade30, 2021-08-29:
Hi,

I am using SX1278 chip from AI-Thinker (433 MHz) with Arduino board to get the temperature and humidity data but I am getting "Starting LoRa failed". I made the changes as mentioned in this thread (for LoRa.cpp and setPins as well) but struggling to debug the issue. Can somebody please help me out here as it is really crucial for me to make it run. The connections and the codes are as follows:

The transmitter code:

#include <SPI.h>

#include <LoRa.h>

#include "DHT.h"

#define DHTPIN A0     // what pin we're connected to

#define DHTTYPE DHT11   // DHT 11
#define LORA_SS 10
#define LORA_RST 9
#define LORA_DIO0 2
DHT dht(DHTPIN, DHTTYPE);

int hum;

float temp; //Stores temperature value

int counter = 0;

void setup() {    
   pinMode(LORA_SS, OUTPUT);
digitalWrite(LORA_SS, HIGH);

LoRa.setPins(LORA_SS, LORA_RST, LORA_DIO0);
LoRa.setSPIFrequency(1E6);

  Serial.begin(9600);

  dht.begin();

  while (!Serial);

  Serial.println("LoRa Sender");
 

  if (!LoRa.begin(433E6)) {

    Serial.println("Starting LoRa failed!");

    while (1);

  }

}

void loop() {

  temp = dht.readTemperature();

  hum = dht.readHumidity();

  Serial.print("Sending packet: ");

  Serial.println(counter);

  // send packet

  LoRa.beginPacket();

  LoRa.print("Humidity: ");

  LoRa.print(hum);

  LoRa.print("c");

  LoRa.print("Temperature:");

  LoRa.print(temp);

  LoRa.endPacket();

  counter++;

  delay(5000);

}

The receiver code:

#include <SPI.h>

#include <LoRa.h>

#include <LiquidCrystal.h>

#define LORA_SS 10
#define LORA_RST 9
#define LORA_DIO0 2 

const int rs = 8, en = 7, d4 = 6, d5 = 5, d6 = 4, d7 = 3; //Mention the pin number for LCD connection

LiquidCrystal lcd(rs, en, d4, d5, d6, d7);//Initialize LCD method

void setup() {
  
    pinMode(LORA_SS, OUTPUT);
digitalWrite(LORA_SS, HIGH);

LoRa.setPins(LORA_SS, LORA_RST, LORA_DIO0);
LoRa.setSPIFrequency(1E6);

  Serial.begin(9600);

  lcd.begin(16, 2);

  while (!Serial);

  Serial.println("LoRa Receiver");

  if (!LoRa.begin(433E6)) {

    Serial.println("Starting LoRa failed!");

    while (1);

  }

}

void loop() {

  // try to parse packet

  int packetSize = LoRa.parsePacket();

  if (packetSize) {

    // received a paket

    Serial.print("Received packet '");

    // read packet

    while (LoRa.available()) {

      char incoming = (char)LoRa.read();

      if (incoming == 'c')

      {

        lcd.setCursor(0, 1);

      }

      else

      {

        lcd.print(incoming);

      }

    }

  }
}

The connections are as follows:
[image]
[image]

@sandeepmistry Please help me out.

Thanks,
Suraj

Kongduino, 2021-08-30:
Please format the code properly (add 4 spaces before every line, or surround the code with 3 backticks ` before the first line and after the last line; also you don't need to skip a line every line...), as it is now it's unreadable... If you're getting a starting failed error, the init code is wrong.

First thing, remove these 2 lines, you have no business messing with the LoRa pins:

      pinMode(LORA_SS, OUTPUT);
      digitalWrite(LORA_SS, HIGH);

Next, reorder your code: one line you do LoRa, then Serial, then lcd, then Serial, then LoRa. Group your code by item:

```c
void setup() {
  Serial.begin(9600);
  while (!Serial);
  lcd.begin(16, 2);
  Serial.println("LoRa Receiver");
  LoRa.setPins(LORA_SS, LORA_RST, LORA_DIO0);
  LoRa.setSPIFrequency(1E6);
  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}
```

If this still doesn't work, you might want to look into `setSPIFrequency`: maybe change the frequency, or, leave it alone altogether.

surajgawade30, 2021-09-01:
@Kongduino Thanks very much for your reply. Actually I tried all the things you mentioned but it's still says the same. I was wondering if I can use 433 MHz freq in Europe. As far as I know, it is valid but still my hardware is not working. Also, I used setPins as it was mentioned on the forum if anyone receives "Starting LoRa failed".

Can you please help me out debugging this?

Thanks

IoTThinks, 2021-09-01:
Remove the LCD code and use standard example code first.

If still hitting error, either wrong pin mapping or not enough power to LoRa.

For less likely issue, you may need to convert logic level to 3.3v for LoRa.

surajgawade30, 2021-09-01:
@IoTThinks Thanks for your reply. Yeah I think there isn't enough power to LoRa. I'll see if an external 3.3V power supply works out or not.

Kongduino, 2021-09-01:
433 is not an issue – at least not a technical one. I am not sure whether it is legal everywhere in the EU but considering most cars/gates remote controls are 433...

As @IoTThinks said, it could be that the LCD is the issue. Remove everything: just the board and LoRa. Also, I am not sure the Ra-01/Ra-02 is 5v tolerant. It's usually rated for 3.3~3.7v. You might've fried it. Do you have a 3.3v board (and a new Ra-0x chip)? Or you could get a bidirectional logic level converter and make sure you protect the LoRa chip.

ASHJAY97, 2021-09-02:
hello, I'm tried to test lora module. Where,on the transmitter side i use arduino uno and lora ra-02. On the receiver side, i use arduino nano and lora ra-02. I'm having issues on the transmitter side which it keep on showing **starting lora failed**. Please help me.

IoTThinks, 2021-09-02:
@ASHJAY97 If you were me, would you know what I did with my boards, my source code and wiring?

So in order to help, you have to take your effort to show your physical boards (to avoid misktakes using wrong boards), your source code (to confirm it is not your coding bugs) and your real wiring (to confirm the wiring is correct).

You must use example codes in this library first.

Kongduino, 2021-09-02:
@IoTThinks I know, I have a few :-) I was being slightly rhetorical. I believe the chip is fried at this point. A sustained application of good ole 5v probably did nothing good to it.

IoTThinks, 2021-09-02:
My ra01/02 can run well with Arduino Mega with 5v logic for a while only.
By right, it is not good.

Most of the time, I use rfm95/ra01/ra02 with esp32 (3.3v logic).

surajgawade30, 2021-09-04:
@sandeepmistry Please do not close this issue yet. I am still working on to resolve the issue.

EsKliwoN, 2022-02-18:
Did you just connect UNO to SX1278 directly? 

If yes, i think maybe that was the problem is

Because, so far i know that LoRa operating voltage is 3V3 which most off all Arduino (UNO, NANO, MEGA) have 5V operating voltage that maybe can burn your LoRa Board.

So, i highly recommend you have to use 3V3 board operating voltage such as ESP32, STM32, etc.

BrunoRobim, 2022-04-08:
I'm having the same problem, but I'm using a esp32 pico with a LoRa embbed to it. The comercial version is: LillyGo TTGO LoRa32 V1.6.1. It was running perfectly, but since my last try it's just doesn't work anymore. And my code is:
`
#include <SPI.h>
#include <LoRa.h>

/*
IMPORTS / INCLUDES
__________
SPI.h   = Interface Serial de Periféricos, permite a comunicação entre
o microcontrolador e outros dipostivos via Serial Port (USB COM do computador)
LoRa.h  = Biblioteca de configurações de firmware para o uso do protocolo de
comunicação LoRa. Configuração de pinagens das placas estão neste documento.

*/

// inicia um contador inteiro iguala a 0
int counter = 0;

// Configuração da inicialização da comunicação LoRa Transmissora
void setup()
{
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Sender");

  if (!LoRa.begin(868E6))
  {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop()
{
  Serial.print("Sending packet: ");
  Serial.println(counter);

  // send packet
  LoRa.beginPacket();
  LoRa.print("hello ");
  LoRa.print(counter);
  LoRa.endPacket();

  counter++;

  delay(1000);
}

`

Kongduino, 2022-04-09:
You're missing two lines I'd think:

```c
  SPI.begin(SCK, MISO, MOSI, SS); // put the right pin numbers or -1 for default. On my TTGO it's 5, 19, 27, 18
  LoRa.setPins(SS, RST, DI0); // on my TTGO it's 18, 14, 26
```

maciek01, 2024-01-11:
adding   SPI.begin(SCK, MISO, MOSI, -1);
fixed my issue (the same error as reported  Starting Lora Failed )

ameykore, 2024-04-03:
Hello Everyone,

Sorry for starting this issue again, but I am facing a similar problem. I am using [this ](https://github.com/sandeepmistry/arduino-LoRa)library to establish loRa communication. 

Hardware:
- Arduino Mega2560 R3
-  LoRa SX1278

Connection:
-  [image]

Following Tutorial:
- [Here](https://how2electronics.com/interfacing-sx1278-lora-module-with-arduino/)

Error:

- Starting LoRa failed!

Solution Tried:
- Changing frequencies
- Example codes than personal code.
- Connected all grounds
- Checked 3V3 connection

The code I am trying is an example sender code on the GitHub link mentioned above.  Also, I removed every extra connection and just kept LoRa and Arduino and tried to send counter values. Of course I changed the code accordingly.

EsKliwoN, 2024-04-04:
@ameykore 

In my opinion, I guess it's because the output signal from the arduino has a value of 5V, while the LoRa works on a 3v3 signal. Not only 3v3 power but also 3v3 signal.

ameykore, 2024-04-04:
@EsKliwoN 

Yes, you are right, I checked the datasheet just now and it does say the operation range is 1.8 - 3.9V. So weird as there are so many tutorials with Arduino and no one mentioned this. 

What is the solution then? Can you suggest me alternative to Arduino? I don't wish to use a converter for each pin.

EsKliwoN, 2024-04-05:
@ameykore 
In my opinion, 
1. use 5V to 3V3 electrical converter
2. Look for a microcontroller with 3v3 standard output such as ESP32, STM32 or others

I dont think have any alternative suggestion to Arduino, except number 1 (use 5V to 3V3 electrical converter)

VikasDange04, 2024-04-11:
hello guys!
I'm facing similar problems related to starting lora failed.In my setup i don't think that theres power supply issue!

setup-
arduino uno
lora sx1278

code-
sender

#include <SPI.h>
#include <LoRa.h>

void setup() {
    Serial.begin(9600);      
    Serial.println("LoRa Sender");
    
    LoRa.setSPIFrequency(1E6);   
    if (!LoRa.begin(433E6)) {
        Serial.println("Starting LoRa failed!");  
        while (1); 
    }    

    LoRa.setTxPower(17);
    LoRa.setSpreadingFactor(12);
    LoRa.setCodingRate4(8);
    LoRa.setSignalBandwidth(250E3);
}

void loop() {
    Serial.println("Sending 'Hello'...");
    LoRa.beginPacket();
    LoRa.print("Hello");
    LoRa.endPacket();
    
    delay(5000);
}

Receiver

#include <SPI.h>
#include <LoRa.h>

void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Receiver");

  LoRa.setSPIFrequency(1E6);

  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }

  LoRa.setTxPower(17); //2-17
  LoRa.setSpreadingFactor(12); //6-12
  LoRa.setCodingRate4(8); //5-8
  LoRa.setSignalBandwidth(250E3);
}

void loop() {
  // try to parse packet
  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    // received a packet
    Serial.println("Received packet");

    // read packet and print it
    while (LoRa.available()) {
      Serial.write(LoRa.read());
    }
    Serial.println(); 
  }
}

please help me out.

VikasDange04, 2024-04-11:
here, one end of lora is working properly means when i try it with sender it sends or the same component with receiver it shows receiving 

but the new purchased lora module new than above does not work on both ends...... shows starting lora failed for both codes....
why?

gianottn, 2024-05-03:
> here, one end of lora is working properly means when i try it with sender it sends or the same component with receiver it shows receiving
> 
> but the new purchased lora module new than above does not work on both ends...... shows starting lora failed for both codes.... why?

I have the same problem, sometimes the module works and sometimes it shows that it failed. I will try to regulate arduino pins from 5V to 3.3V otherwise I have no idea why it does not work

suzu2004, 2025-03-04:
im using lorawan ra-02 with esp 8266 node mcu and oled but lora isnt getting initialised pls help

EsKliwoN, 2025-11-14:
> > [@EsKliwoN](https://github.com/EsKliwoN)
> > Yes, you are right, I checked the datasheet just now and it does say the operation range is 1.8 - 3.9V. So weird as there are so many tutorials with Arduino and no one mentioned this.
> > What is the solution then? Can you suggest me alternative to Arduino? I don't wish to use a converter for each pin.
> 
> did anyone find a solution for this?

@Nomunomm I suggest to use 3v3 basic microcontroller power and signal
Otherwise, use the signal converter from Power Source (example 5V like arduino) to 3V3 signal