sandeepmistry/arduino-LoRa #511: LoRa.begin fails

bakwc, 2021-09-20:
Hi. How can I debug LoRa.begin returning false?
I've checked:
1) I use 433E6 frequency, I have sx1278 that support this frequency
2) I set correct pins (LoRa.setPins(PA_4, PB_0, PB_1)), PB_1 is not connected as it mentioned as optional
3) I tried to set LoRa.setSPIFrequency(1E6); Also I tried without setting any frequency
4) I double-checked my wiring and checked different modules

Here is my code:

```

void setup() {
    pinMode(LED_BUILTIN, OUTPUT);
    serial.begin(115200);

    delay(5000);
    serial.write("starting lora\n");
    delay(1000);

    LoRa.setSPIFrequency(1E6);

    serial.write("frequency set\n");
    delay(1000);

    LoRa.setPins(PA_4, PB_0, PB_1);

    serial.write("pins set\n");
    delay(1000);

    if (!LoRa.begin(433E6)) {
        Serial.println("Starting LoRa failed!");
        fail();
    }

    serial.write("started\n");
}

```

Kongduino, 2021-09-20:
`LoRa.begin()` returns `false` only in one case – when, after starting SPI, it tries to read the `REG_VERSION` register, and compares it with the expected result, `0x12`. So something is screwy with your SPI settings. It could be that your wiring is wrong, or that `LoRa.setSPIFrequency(1E6);` sets the SPI frequency too high. Try again without that line.

`LoRa.setPins()` sets ss, reset, and dio0 in that order. DIO0 is indeed optional (it's the pin that signals there's activity, used for asynchronous communication). Check SS and RESET are in the right order.

That's the only things I can think of.

bakwc, 2021-09-20:
I checked `readRegister` - it returns 255 (when I power off module - it returns 0).

IoTThinks, 2021-09-20:
Are you using stm32 + stm32duino + this library?

bakwc, 2021-09-20:
Yes, I'm using stm32 blue pill.
I tried to set pins and initialize in a while loop, but now my board just stop responsing after the first failure.

IoTThinks, 2021-09-20:
SPI.setMISO(LORA_MISO);
SPI.setMOSI(LORA_MOSI);
SPI.setSCLK(LORA_SCK);
SPI.setSSEL(LORA_NSS);
  
LoRa.setPins(LORA_NSS, LORA_NRESET, LORA_DIO0);

bakwc, 2021-09-20:
For me this worked:
https://pastebin.com/1SNSH4bE

The output is:

```
18:28:37.041 -> starting lora
18:28:38.038 -> frequency set
18:28:39.042 -> pins set
18:28:40.038 -> started
18:28:41.069 -> init failed
18:28:42.071 -> init failed
18:28:43.067 -> init failed
18:28:44.071 -> init failed
18:28:45.074 -> init failed
18:28:46.044 -> init failed
18:28:47.049 -> init failed
18:28:48.054 -> init failed
18:28:49.054 -> init failed
18:28:50.054 -> init failed
18:28:51.051 -> init failed
18:28:52.059 -> success init
18:28:53.062 -> success init
18:28:54.093 -> success init
18:28:55.069 -> success init
18:28:56.069 -> success init
```

closed 2021-09-20 by bakwc