sandeepmistry/arduino-LoRa #512: LoRa.read() stuck

bakwc, 2021-09-20:
1) When trying to receive packet with:

```
            serial.printf("Packet received: %d\n", packetSize);
            while (LoRa.available()) {
                serial.write(LoRa.read());
            }
```
Everything stucks.

2) LoRa.parsePacket() returns non-zero even if I didn't send anything.

3) If I ignore LoRa.parsePacket for the fist time - it does not trigger any times more, no packet received at all.

The whole code: https://pastebin.com/qqtRUREZ

Kongduino, 2021-09-20:
I had a look at your code, and you have this line:

```c
  LoRa.setPins(PA_4, -1, -1);
```

This is the issue I think.

The prototype for `setPins` is:

```c
void LoRaClass::setPins(int ss, int reset, int dio0)
```

While the third pin, DIO0, is optional, the second pin, RESET, is compulsory. The `begin()` function starts with a reset sequence. Without a reset pin, you won't be able to make the chip work. Connect the LoRa chip's RESET pin to a pin on your MCU.

IoTThinks, 2021-09-21:
I guess there will be a question about high packet loss in receiver node.

bakwc, 2021-09-21:
> I guess there will be a question about high packet loss in receiver node.

Sort of. After adding reset connection it doesn't stuck anymore, but there is no received packets at all.

```
12:37:50.427 -> starting lora
12:37:51.437 -> frequency set
12:37:52.420 -> pins set
12:37:53.453 -> started
12:37:54.448 -> init failed
12:37:55.458 -> init failed
12:37:55.494 -> success init
12:37:55.494 -> no packet
12:37:56.483 -> success init
12:37:56.483 -> no packet
12:37:57.490 -> success init
12:37:57.490 -> no packet
12:37:58.508 -> success init
12:37:58.508 -> no packet
12:37:59.512 -> success init
12:37:59.512 -> no packet
12:38:00.520 -> success init
12:38:00.520 -> no packet
12:38:01.494 -> success init
12:38:01.494 -> no packet
12:38:02.521 -> success init
12:38:02.521 -> no packet
...
```
Not a single packet.

Kongduino, 2021-09-21:
Your LoRa init code should stay in `setup()`. Code in `loop()` is repeated continuously and is for when things have been set up (*chuckle*) and are working smoothly. Put the init code within a `while(!initialized){}`. Meanwhile, `loop()` should only have the `parsePacket` code.

bakwc, 2021-09-21:
Moved code to setup. Here is the code: https://pastebin.com/C4K8VWuk
The output on the receiver side:
```
13:38:15.571 -> starting lora
13:38:54.571 -> frequency set
13:38:55.559 -> pins set
13:38:56.573 -> initializing lora, attempt 1
13:38:57.602 -> started
13:38:58.615 -> Packet received: 1
13:38:58.615 -> 
```
And everything stuck.

Kongduino, 2021-09-22:
@IoTThinks's comment about high packet loss was maybe a little cryptic, but not random. In your code you added the transmitter code, commented out, for reference. Of course since it's an excerpt, we have no way of knowing how often you send a packet. But if there's no delay, and you keep pumping out packets, yeah, you have another problem on your hands...

Put a delay, maybe 10 seconds, between packets. See if that improves things on the receiver side.

```c
  // ====== TRANSMITTER CODE ======
  //    LoRa.beginPacket();
  //    LoRa.print("hello ");
  //    LoRa.print(counter++);
  //    LoRa.endPacket();
  //    serial.write("packet transmitted\n");
```

bakwc, 2021-09-22:
I already have 1 second delay:
```
    digitalWrite(LED_BUILTIN, HIGH);
    delay(500);
    digitalWrite(LED_BUILTIN, LOW);
    delay(500);
```
I have the problem with receiver, not the transmitter. It stucks or receive no packets at all.

EsKliwoN, 2022-01-28:
I think i have solved your problem @bakwc with my code here

https://pastebin.com/F9jgeRhe

so, you have to put delay when it receive data. Eventhough the delay is 100ms.
Actually, i have tried to put 25ms delay but the data unconstant.
Hope it solved ✌️