sandeepmistry/arduino-LoRa #514: CAD - Channel free detection?

bobricius, 2021-09-29:
Hi, Is there any way how to detect if is not transmission on channel?

Kongduino, 2021-09-29:
Yes. You need to set up `RegIrqFlagsMask`'s (0x11) bit 0 – `CadDetectedMask` – to set up Channel Activity Detection, and use `RegIrqFlags`'s (0x12) bits 0 (`CadDetected`) and 2 (`CadDone`) to decide whether the channel is free to use or not.

LoRa.h has the two bit masks defined.

```c
#define IRQ_CAD_DONE_MASK          0b00000100
#define IRQ_CAD_DETECTED_MASK      0b00000001
```
`CadDetected` is on DIO4 and `CadDone` on DIO3 in DIOx Mapping `00`, DIO1 and DIOØ in DIOx Mapping `10`. This is configured with registers `RegDioMapping1` and `RegDioMapping2` (0x40/0x41).

But there is no special provision for sending when the channel is free, like in Semtech's sx1262 library. Maybe a feature request...? This is a bit complex to set up properly.

halukmy, 2023-02-15:
any update @Kongduino ?

Kongduino, 2023-02-15:
> any update @Kongduino ?

I wasn't aware I was supposed to follow up on this... My answer is as complete as can be, given the circumstances.

halukmy, 2023-02-15:
has got any other alternatives?

morganrallen, 2023-02-15:
There are two open pull requests, if someone tests them and let's me know they work I'll merge and do a release.

bobricius, 2023-02-15:
I am implemented function from this PR ... [https://github.com/sandeepmistry/arduino-LoRa/pull/574](https://github.com/sandeepmistry/arduino-LoRa/pull/574) and work beautiful. It is logical ..."if receiving something = somebody transmitting" CAD is is almost useless, this status bit is documented in datasheet with les than one sentence

morganrallen, 2023-02-15:
That PR isn't implementing true CAD, you have to write to the `OpMode` register to enable CAD and `DioMapping1` register if you want an interrupt, as CAD is an asynchronous operation.

If I understand the `ModemStatus` register correctly, this is when the modem is in RX mode (Single, or Continuous).

So it will only detect when YOU are receiving, full CAD will detect if two OTHER devices are talking on your channel.

This functionality does look useful, but I feel like the PR is lacking, would be nice to implement all of the `ModemStatus` at the same time BUT, if someone follows up adding this to `API.md` I'll merge as is.

As for true `CAD`, PR #334 looks fairly complete, it needs some very minor tweaking and validation that it works and it can be merged.

halukmy, 2023-02-16:
Thanks @bobricius  and @morganrallen 

can you inform us more about CAD?

is it kind of WiFi style communication? like all time mesh connected? glad to be know-how and a detail if possible

and thanks for being update and make it process we are looking forward for update

Kongduino, 2023-02-16:
Here's [an article about CAD](https://lora-developers.semtech.com/documentation/tech-papers-and-guides/channel-activity-detection-ensuring-your-lora-packets-are-sent/how-to-ensure-your-lora-packets-are-sent-properly) from the Semtech website. It's quite technical but explains CAD quite well – and it's from the horse's mouth, so to speak.