sandeepmistry/arduino-LoRa #518: problems with lora

Arlencha, 2021-10-05:
I have some problems with lora, I use arduino and I put the code LoReceiver and LoraSender from arduino but it doesn't receive any data. I would like to know how I can fix it :(

IoTThinks, 2021-10-06:
Show your physical boards, real pin mapping and real code.

Arlencha, 2021-10-06:
[image]
I use sx1276 module
pin:
               ------------------------------
GND    -                                               -  ANT
     X     -                                               -  GND
REST    -                                               -   X
NSS     -                                               -   X
SCK     -                                               -   VCC
MOSI   -                                               -   DI00
MISO   -                                               -   X
 GND   -                                               -   X
               -------------------------------
code transmitter:
#include <SPI.h>
#include <LoRa.h>
int pot = A2;
 
void setup() {
  Serial.begin(9600);
  pinMode(pot,INPUT);
  
  while (!Serial);  
  Serial.println("LoRa Sender");
  if (!LoRa.begin(433E6)) { // or 915E6, the MHz speed of your module
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}
 
void loop() {
  int val = 365;
  LoRa.beginPacket();  
  LoRa.print(val);
  LoRa.endPacket();
  delay(50);
 
}

code reciver:
#include <SPI.h>
#include <LoRa.h>  
String inString = "";    // string to hold input
int val = 0;
 
void setup() {
  Serial.begin(9600);
  
  while (!Serial);
  Serial.println("LoRa Receiver");
  if (!LoRa.begin(433E6)) { // or 915E6
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}
 
void loop() {
  
  // try to parse packet
  int packetSize = LoRa.parsePacket();
  if (packetSize) { 
    // read packet    
    while (LoRa.available())
    {
      int inChar = LoRa.read();
      inString += (char)inChar;
      val = inString.toInt();       
    }
    inString = "";     
    LoRa.packetRssi();    
  }
  Serial.print("POT:");    
  Serial.println(val);  
 delay(1000);
}

IoTThinks, 2021-10-13:
This is ra02 sx1278.
Can send at 433Mhz.

Arlencha, 2021-10-13:
but I'll change the frequency and then show what happend :)

helia77, 2021-11-04:
Did you solve the problem?
These options might be helpful:
1. The serial monitor is also set to 9600 in the receiver page.
2. Which region are you in? 433 is for Asia for example. Make sure you set it to your own region.
3. Check the wirings again. There might be a damaged wire.
4. I am not sure if you can use the exact same wiring as for the sx1278 RA-02 Lora module.
5. Have you been using an antenna from the first place? Because you haven't, there might be a slightly chance of your Lora module is burnt. (happened to me)
6. Your delay is too short. Make it to 5000 (5 secs.)

and one more question, do you see anything in your serial monitor for the receiver? The "Starting Lora failed!" message?

Arlencha, 2021-11-09:
1. yes
2.  In mexico; I know is 915
3. I change all wiring and I put a voltage divider, but now It doesn't work,
4. I'm using sx1278
I don't know what I can do it, help plis :'(
I use this
Semtech SX1276/77/78/79 | Arduino
pinLoRa | pin arduino
VCC | 3.3V
GND | GND
SCK | SCK
MISO | MISO
MOSI | MOSI
NSS | 10
NRESET | 9
DIO0 | 2

Arlencha, 2021-11-09:
Or Do you How to connect Lora module to Arduino?  I don't know I'm doit well :(

Arlencha, 2021-11-10:
I fix it! I just need a voltage divider in the input of MOSI, SCK y Nss!
pinLoRa | pin arduino
VCC ----|----3.3V
GND ---|--- GND
SCK---- |--- SCK (with voltage divider) 
MISO --|--- MISO (It doesn't use voltage divider)
MOSI --|--- MOSI (with voltage divider) 
NSS ---|---- 10 (with voltage divider) 
I used 1k and 2K for resistors
thanks at all! :)

closed 2021-11-10 by Arlencha

IoTThinks, 2021-11-10:
@Arlencha Would you mind sharing the voltage devider (model, photo, link...)?

So other people can learn from it too? Thanks a lot.

uknowmeset, 2024-01-15:
Please I need help also regarding that problem. Can you show me the voltage divider that you created?. I would like to clarify, where did you put the voltage divider, in the Rx or Tx?

4lsdima, 2025-09-22:
> I fix it! I just need a voltage divider in the input of MOSI, SCK y Nss! pinLoRa | pin arduino VCC ----|----3.3V GND ---|--- GND SCK---- |--- SCK (with voltage divider) MISO --|--- MISO (It doesn't [...]

Спасибо! Мне помог именно ваш способ! Я сделал точно также, и все заработало!