sandeepmistry/arduino-LoRa #529: Arduino Uno SX1278 Starting LoRa failed!

saifularifinww, 2021-11-08:
Please help, I had problem "Starting LoRa failed!"

Device : 
Arduino Uno
LoRA-02, SX 1278,  433 MHz

I'm using example code LoRaSender
    
    #include <SPI.h>
    #include <LoRa.h>
    
    int counter = 0;
    
    void setup() {
      Serial.begin(9600);
      while (!Serial);
    
      Serial.println("LoRa Sender");
    
      if (!LoRa.begin(433E6)) {
        Serial.println("Starting LoRa failed!");
        while (1);
      }
    }
    
    void loop() {
      Serial.print("Sending packet: ");
      Serial.println(counter);
    
      // send packet
      LoRa.beginPacket();
      LoRa.print("hello ");
      LoRa.print(counter);
      LoRa.endPacket();
    
      counter++;
    
      delay(5000);
    }

Wiring Schematic:
[image]

<html>
<body>

SX1278 | Arduino Uno | Power Supply 3.3 Volt
-- | -- | --
VCC |  | +
GND | GND | -
SCK | SCK (Pin 13)  | 
MISO | MISO (Pin 12) | 
MOSI | MOSI(Pin 11) | 
NSS | Pin 10 | 
NRESET | Pin 9 | 
DIO0 | Pin 2 | 

</body>
</html>

I already using external power supply, and try add code setPins

    #define LORA_SS 10
    #define LORA_RST 9
    #define LORA_DIO0 2

and

    pinMode(LORA_SS, OUTPUT);
    digitalWrite(LORA_SS, HIGH);
    
    LoRa.setPins(LORA_SS, LORA_RST, LORA_DIO0)

I also try change SPI Frequency using this code

      LoRa.setSPIFrequency(4E6);
But still didn't work. Please anyone help me about this issues. Thank you

Kongduino, 2021-11-08:
It seems we have the same question popping up regularly here.1. Leave the pins alone. Do not try to set them up yourself.2. Your Arduino Uno is 5v. Your sx1278 is 3.3v. It is probably fried by now. Use a board that works on 3.3v. An esp32, esp8266, etc.Sent from my Galaxy

saifularifinww, 2021-11-08:
Thanks for your reply

I don't really understand,
Does it means LoRa SX1278 not compatible with Arduino Uno? Because I found a lot example using Uno or I missing some module/part that I didn't know?

and did my board already broken because my previous wiring/connection?

Kongduino, 2021-11-08:
Hi again.

It's not so much a problem of compatibility with the Arduino Uno itself – it's the 5V. I have a Chinese Arduino Uno clone that can be switched to 3.3v. It works perfectly with a Ra-02. But running a Ra-02 or similar on 5v for any length of time is going to kill your chip, as it's max tolerance is (if memory serves) 3.6V.

You can either use a 3.3v board, as I said, or put a bidirectional Logic Level Converter between the Uno and the LoRa chip. If you have such a board, test the chip on it. It could be dead, or still functioning. My bet's on dead but who knows...