sandeepmistry/arduino-LoRa #530: SPI half duplex pins

CodeOn-ArK, 2021-11-10:
Hello!
I don't have any issue, rather a question - the sx1278 SOC uses half duplex mode of communication so why are we using all the SPI pins? As far as I know to communicate in Half Duplex mode you need only one pin as shown here -
```shorturl.at/ivyzB```
So why are we using 4 pins?
Thanks in advance

CodeOn-ArK, 2021-11-10:
> Interesting question.
> 
> May be Semtech want to use I2C standard instead of reinventing the wheel?

wdym? Doesn't it uses SPI by default and is the only means of communication?

IoTThinks, 2021-11-10:
@CodeOn-ArK My bad. SPI.
Already edited. Thanks a lot.

closed 2022-03-02 by CodeOn-ArK

morganrallen, 2022-03-02:
Not sure where this info is coming from but far as I can tell the SX127x series are full-duplex only.

[image]