sandeepmistry/arduino-LoRa #532: Implicit header mode working - I can not get rid of the header

foorschtbar, 2021-11-17:
I can't get the implicit header mode working. All transmissions are in the explicit mode with header. Issue #382 doesnt help me.

I record the LoRa traffic with a GNURadio setup and visualize it in audiacity. 

```
  LoRa.beginPacket(true);
  LoRa.write(0x80);
  LoRa.write(0x80);
  LoRa.write(0x80);
  LoRa.write(0x80);
  LoRa.endPacket();
```

I use SF8, CR4/8 and no CRC which results in 8 symbols for 4 byte of payload. But i can see 16 symbols:

[image]

When i extend the paylod to 6 or 8 bytes, i got 20 or 24 symbols where are still 8 symbols to much.

https://github.com/sandeepmistry/arduino-LoRa/blob/01bf044b769a2ab276b59fa8cb24e587b24fc787/src/LoRa.cpp#L685-L690

The code from the libary looks good. The bit 0 in the 0x1d register (RegModemConfig 1) is set to 1, which is for the implicit header mode. i have no idea left why it is not working :-(

IoTThinks, 2021-11-18:
"Not working" means can still send/receive in implicit mode but the header is still there?
Could you make the payload a bit longer like 16 and 32 bytes and check the total lenght of the packet? To guess why the extra symbols come from.

Kongduino, 2021-11-18:
What's your preamble length? It's 8 by default, which would account for the extra 8 bytes.

foorschtbar, 2021-11-18:
> "Not working" means can still send/receive in implicit mode but the header is still there?
> 
> Could you make the payload a bit longer like 16 and 32 bytes and check the total length of the packet? To guess why the extra symbols come from.

I have change the length and see the symbols for each extra byte. 4 more bytes means 8 extra symbols.

foorschtbar, 2021-11-18:
> What's your preamble length? It's 8 by default, which would account for the extra 8 bytes.

The default 8. Thats the first 8 symbol in the image. The issue is the payload part which is too long. When i switch between implicit and explicit mode nothing - exactly nothing change in the transmitted output.

Sirashs, 2022-12-01:
hello, Foorschtbar, Have you solved the problem?
[image]

foorschtbar, 2022-12-01:
Hi, my problem was not related to a compiling issue, sry :(

kt-work, 2022-12-09:
> hello, Foorschtbar, Have you solved the problem? I need some help [image]

You cannot call it directly, it is called in beginPacket()
[image]

kt-work, 2022-12-09:
> When i extend the paylod to 6 or 8 bytes, i got 20 or 24 symbols where are still 8 symbols to much.
> 
This sounds like a problem Matt Knight of Bastille research reported in one of his presentations. In implicit mode the header SF and CR remain the same as in explicit mode. Maybe the gr-lora code is worth checking for how this is handled.