sandeepmistry/arduino-LoRa #533: Recommended heigh for devices

Nehuix, 2021-11-22:
Hi guys!
I am developing a tracker for animals using Arduino nano and sx1278 LoRa. I am not having a very large transmitting range (around 200m at 4m height) and I would like to know any recommendations for the height that my reciever (base) should be since my transmitter can't be any hight than an average cow. Maybe use a 3dBi antenna since I am using the little spiral one.
Thanks in advance!

Kongduino, 2021-11-22:
What are the radio settings of the SX1278? Frequency, SF/BW etc... These are important.

IoTThinks, 2021-11-22:
Try to use external 3.3v power for LoRa.

Nano is known for lacking of power output for LoRa (up to 100-120mA)

Nehuix, 2021-11-25:
> What are the radio settings of the SX1278? Frequency, SF/BW etc... These are important.

They are all set by default so I don't really know, sorry