sandeepmistry/arduino-LoRa #535: Waking up arduino on receiving data

ShadowKing782k, 2021-11-29:
I want to put my arduino in sleep mode, it should wake up on receiving data from receiver LoRa. Is it possible to do that??..If it's possible
can u please tell me the  procedure to do that...Thanks in Advance : )

IoTThinks, 2021-12-06:
1. Use onReceive() ISR
2. Leave your LoRa in Receive mode.
3. When there is some LoRa message, DIO0 will go HIGH.
4. You have to set DIO0 to wake up the Arduino.

BTW, LoRa in Receive mode still consume battery. So be aware of that.

hoseinaghajari, 2022-02-26:
> 1. Use onReceive() ISR
> 2. Leave your LoRa in Receive mode.
> 3. When there is some LoRa message, DIO0 will go HIGH.
> 4. You have to set DIO0 to wake up the Arduino.
> 
> BTW, LoRa in Receive mode still consume battery. So be aware of that.

I tested that a lot but I've found that when we put Arduino to sleep mode, LoRa module can't receive or sent anymore. so from my point of view, LoRa and its DIO pins can't do that for us. did you tested that?

wedevelper, 2023-10-11:
I am having the same issue. I am using ESP32 and when i put it in light sleep mode, LoRa module sx1278 then stops receiving any data. When it is normally operating the DIO0 goes high till i read it and goes low after reading data. But in sleep mode it goes to low. I want to wake up my ESP32 when there is something received on LoRa module. If anyone has tried and got the results please let me know.

LooOnGit, 2024-07-31:
> I am having the same issue. I am using ESP32 and when i put it in light sleep mode, LoRa module sx1278 then stops receiving any data. When it is normally operating the DIO0 goes high till i read it [...]

can you share i also have problem with DIO0 pin

martinius96, 2025-05-12:
For all that are using ESP32 and would like to wake it up from light sleep via DIO0, here is the solution that works:

```
void setup() {
  Serial.begin(115200);
  LoRa.setPins(SS, RST, DI0);
  if (!LoRa.begin(433E6)) {
    while (true);
  }
  LoRa.receive();
  delay(10);
  pinMode(RST, INPUT_PULLUP);
  gpio_wakeup_enable(GPIO_NUM_2, GPIO_INTR_HIGH_LEVEL); //in my case GPIO_NUM2 is DIO0
  esp_sleep_enable_gpio_wakeup();
}

void loop() {
  esp_light_sleep_start();
  Serial.println("Incoming Message");

  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    Serial.print("Received packet '"); // read packet
    while (LoRa.available()) {
      Serial.print((char)LoRa.read());
    }
  }
  LoRa.receive();
  delay(10);
}
```