sandeepmistry/arduino-LoRa #536: Problem with LoRa onTxDone and SSD1306Lib

Baptou88, 2021-12-01:
Hello Everyone !

I just noticed a problem with this library and some libraries for the use of an ssd1306 and I would like to understand where this problem comes from and how it is possible to fix it

here is my situation: 

- I use an ESP32 with OLED 0.96" by Heltec ([this product](https://heltec.org/project/wifi-lora-32/)
- I use this lib to use LoRa functionality (I'm not use the [official lib](https://github.com/HelTecAutomation/Heltec_ESP32) because they don't implement the onTxDone function)

At this point all is Ok my project work correctly

However when I want to use the screen through these libs (Adafruit_SSD1306 or [this one](https://github.com/ThingPulse/esp8266-oled-ssd1306) ) I'have an error on The Serial Monitor `[E][esp32-hal-i2c.c:1434] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=1, scl=1` which occurs just after passing the function 'onTxDone'

`#include <Arduino.h>
#include <SPI.h>              // include libraries
#include <LoRa.h>
// #include <Adafruit_GFX.h>
// #include <Adafruit_SSD1306.h>
#include <SSD1306Wire.h>
#include <Wire.h>

const long frequency = 868E6;  // LoRa Frequency

const int csPin = 10;          // LoRa radio chip select
const int resetPin = 9;        // LoRa radio reset
const int irqPin = 2;          // change for your board; must be a hardware interrupt pin

#define SCREEN_ADDRESS 0x3C

//Adafruit_SSD1306 display(128  , 64, &Wire1, RST_OLED);
//Adafruit_SSD1306 display(4);

SSD1306Wire display(0x3c,SDA_OLED,SCL_OLED,GEOMETRY_128_64);

void onReceive(int packetSize){
  Serial.println("I've receive a packet");
}
void onTxDone(){
  Serial.println("txDone " + String( millis()));
  LoRa.receive();
}
void setup() {
  // put your setup code here, to run once:
 
  Serial.begin(9600);
  

  
//  if(!display.begin(SSD1306_SWITCHCAPVCC, 0x3c)) { //SSD1306_EXTERNALVCC
//     Serial.println(F("SSD1306 allocation failed"));
//     for(;;); // Don't proceed, loop forever
//   }
    pinMode(16, OUTPUT);
    digitalWrite(16, HIGH);
    delay(10);

    if (!display.init())
    {
    Serial.println("Error init display");
    } 

  

    display.drawCircle(10,10,5);
    display.display();

    SPI.begin(SCK,MISO,MOSI,SS);
    LoRa.setPins(18,14,26);
    pinMode(25,OUTPUT);
    while (!LoRa.begin(frequency)) {
        Serial.println("Starting LoRa failed!");
        delay(500);
    
    }
    Serial.println("Lora ok");

    LoRa.onReceive(onReceive);
    LoRa.onTxDone(onTxDone);
    LoRa.receive();
    delay(1000);
}

void loop() {
  // put your main code here, to run repeatedly:
  // display.clearDisplay();
  // display.setCursor(10,10);
  // display.println("Bonjour ");
  // display.drawCircle(20,20,10,SSD1306_WHITE);
  static int test = 0;
  display.clear();
  if (test > 128)
  {
    test = 0;
  }
  test++;
  display.setPixel(test,10);
  display.display();
  static unsigned long lastsend =0;
  if (millis()-lastsend > 5000) //send packet every 5sec
  {
    lastsend = millis();

    LoRa.beginPacket();
    LoRa.println("bonjour");

    digitalWrite(25,HIGH);
    LoRa.endPacket(true);

    //LoRa.receive();
    Serial.println(millis());
    digitalWrite(25,LOW);
    
  }
 
 
}

`

I hope I have been understandable enough

Kongduino, 2021-12-02:
Please format your code properly. Not 1 ` but 3, before and after the code. Like this:

```
void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

Baptou88, 2021-12-02:
thanks @Kongduino for the the reply

Kongduino, 2021-12-03:
First thing I can is this: `SDA_OLED` and `SCL_OLED` are not defined anywhere, and the error message is complaining about `Can't init sda=1, scl=1`. Can you try the init code with actual values?

`SSD1306Wire display(0x3c, <PIN NUMBER>, <PIN NUMBER>, GEOMETRY_128_64);`

Baptou88, 2021-12-03:
Effectively I've forgot to mention sda_oled and scl_oled

I use [this board](https://resource.heltec.cn/download/WiFi_LoRa_32/WIFI_LoRa_32_V2.pdf) so this 2pin are declared in the arduino.pin (pin 15 and 4)

Kongduino, 2021-12-03:
Indeed I have the same boards, bought them a long time ago. So I went dumpster diving and found some old code. I init the OLED like this:

```c
SSD1306  display(0x3c, 4, 15);

void setup() {
[...]
  digitalWrite(16, HIGH);
  // while OLED is running, must set GPIO16 in high
  display.init();
  display.flipScreenVertically();
  display.setFont(ArialMT_Plain_10);
  display.drawString(0, 0, "\n\nOLED Inited");
  Serial.println("\n\nOLED Inited\nReceiver");
  SPI.begin(5, 19, 27, 18);
  LoRa.setPins(18, 14, 23);
  if (!LoRa.begin(433e6)) {
    Serial.println("Starting LoRa failed!");
    display.drawString(0, 20, "LoRa Init Fail");
    display.display();
    while (1);
  }
```

Try this init code, with numbers and not defines, see if it works first.

Baptou88, 2021-12-03:
thanks for your reply !

I've try:

1. replaced `SSD1306Wire display(0x3c,SDA_OLED,SCL_OLED,GEOMETRY_128_64);` by `SSD1306Wire display(0x3c,4,15,GEOMETRY_128_64,I2C_TWO);`

2. replaced `LoRa.setPins(18, 14, 26);` by  `LoRa.setPins(18, 14, 23);`

results are the same:
screen display correctly during ~30/60sec, after i've this msg on the serial monitor `[E][esp32-hal-i2c.c:1434] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=1, scl=1` and randomly a reboot

Baptou88, 2021-12-04:
`Backtrace: 0x4008587c:0x3ffbea70 0x40085af1:0x3ffbea90 0x40086867:0x3ffbeab0 0x400d4059:0x3ffbeaf0 0x400d157d:0x3ffbeb10 0x400d16b0:0x3ffbeb40 0x400d16e4:0x3ffbeb60 0x400d1aa1:0x3ffbeb80 0x40080f1e:0x3ffbeba0 0x40080f69:0x3ffbebc0 0x40084139:0x3ffbebe0 0x40088423:0x3ffb1e20 0x400870ff:0x3ffb1e40 0x400d374e:0x3ffb1e70 0x400d3af3:0x3ffb1ea0 0x400d1c71:0x3ffb1ec0 0x400d1ca5:0x3ffb1ee0 0x400d1d01:0x3ffb1f00 0x400d0e53:0x3ffb1f20 0x400d1002:0x3ffb1f40 0x400d131d:0x3ffb1f70 0x400d4ce9:0x3ffb1fb0 0x40086b01:0x3ffb1fd0
  #0  0x4008587c:0x3ffbea70 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #1  0x40085af1:0x3ffbea90 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #2  0x40086867:0x3ffbeab0 in xQueueGenericReceive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:2038
  #3  0x400d4059:0x3ffbeaf0 in spiTransaction at C:\Users\Baptou\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-spi.c:283
  #4  0x400d157d:0x3ffbeb10 in SPIClass::beginTransaction(SPISettings) at C:\Users\Baptou\.platformio\packages\framework-arduinoespressif32\libraries\SPI\src/SPI.cpp:269
  #5  0x400d16b0:0x3ffbeb40 in LoRaClass::singleTransfer(unsigned char, unsigned char) at .pio\libdeps\heltec_wifi_lora_32\LoRa\src/LoRa.cpp:730
  #6  0x400d16e4:0x3ffbeb60 in LoRaClass::readRegister(unsigned char) at .pio\libdeps\heltec_wifi_lora_32\LoRa\src/LoRa.cpp:730
  #7  0x400d1aa1:0x3ffbeb80 in LoRaClass::handleDio0Rise() at .pio\libdeps\heltec_wifi_lora_32\LoRa\src/LoRa.cpp:694
  #8  0x40080f1e:0x3ffbeba0 in LoRaClass::onDio0Rise() at .pio\libdeps\heltec_wifi_lora_32\LoRa\src/LoRa.cpp:730
  #9  0x40080f69:0x3ffbebc0 in __onPinInterrupt at C:\Users\Baptou\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-gpio.c:274
  #10 0x40084139:0x3ffbebe0 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
  #11 0x40088423:0x3ffb1e20 in xTaskResumeAll at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
  #12 0x400870ff:0x3ffb1e40 in xEventGroupWaitBits at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/event_groups.c:338
  #13 0x400d374e:0x3ffb1e70 in i2cProcQueue at C:\Users\Baptou\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-i2c.c:1287
  #14 0x400d3af3:0x3ffb1ea0 in i2cWrite at C:\Users\Baptou\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-i2c.c:1629
  #15 0x400d1c71:0x3ffb1ec0 in TwoWire::writeTransmission(unsigned short, unsigned char*, unsigned short, bool) at C:\Users\Baptou\.platformio\packages\framework-arduinoespressif32\libraries\Wire\src/Wire.cpp:161
  #16 0x400d1ca5:0x3ffb1ee0 in TwoWire::endTransmission(bool) at C:\Users\Baptou\.platformio\packages\framework-arduinoespressif32\libraries\Wire\src/Wire.cpp:161
  #17 0x400d1d01:0x3ffb1f00 in TwoWire::endTransmission() at C:\Users\Baptou\.platformio\packages\framework-arduinoespressif32\libraries\Wire\src/Wire.cpp:161
  #18 0x400d0e53:0x3ffb1f20 in SSD1306Wire::sendCommand(unsigned char) at src/main.cpp:106
  #19 0x400d1002:0x3ffb1f40 in SSD1306Wire::display() at src/main.cpp:106
  #20 0x400d131d:0x3ffb1f70 in loop() at src/main.cpp:122
  #21 0x400d4ce9:0x3ffb1fb0 in loopTask(void*) at C:\Users\Baptou\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:23
  #22 0x40086b01:0x3ffb1fd0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)`

IoTThinks, 2021-12-06:
It seems that you are trying to display something on OLED (error line 18) while receiving data in LoRa (error line 8).
If yes, then don't do so.

In on the onReceive(), can try to set a flag or a string only.
Then inside the loop, try to detect the flag and display a string in OLED.

onReceive() is an ISR, not a normal method. Try to avoid doing stuffs inside an ISR.