sandeepmistry/arduino-LoRa #539: reset LoRa chip

max763, 2021-12-06:
Hi, I'm using the RFM95W together with an ESP32 to send and receive data via LoRa. Everything is working fine but once in a while it stops working and the only thing that helps is to turn it off an on. Is there a known way to reset the chip in such cases?
`digitalWrite(_reset, LOW);
      delay(10);
      digitalWrite(_reset, HIGH);
      delay(10);`     
doesn't help.
thanks a lot
max

morganrallen, 2021-12-06:
Toggling the reset pin will be the radio back up in an uninitialized state. You'll need to run `LoRa.begin` again. Off hand I cannot think if this will have any side effects.

max763, 2021-12-07:
thanks, will try that! And is there a way to find out whether the chip is in a state that needs a LoRa.begin() to work again?