sandeepmistry/arduino-LoRa #541: How to set up PA_Output to meet restrictions by local law (max 10 mW ERP)

ChangeD20, 2021-12-17:
I´m using Heltec Wifi LoRa 32 v2 (SX1278) at 433mHz and external factory antenna to send data between devices in short range (<100m). 

As I could find out the max transmission power has to be below 10 mW ERP in Germany to use it without paying for permission.

How should I  configurate the lora module to meet the regulations?

Is PA_Output the key to succes? What effect makes the antenna?

Has the spreading factor any effect on it?

How to calculate it?

Please help.

IoTThinks, 2021-12-18:
Check for TX power in the API.

Need to convert mW to dB.

Kongduino, 2021-12-18:
According to [this calculator](http://rfcalculator.mobi/convert-erp-mw.html), 10 ERP mW is 10 ERP dBm. Use the `setTxPower(int level, int outputPin)` function to do that.

With outputPin = `PA_OUTPUT_RFO_PIN` (ie 0) you get:

[image]

`MaxPower` is set in this scenario at 7 by the library:

`writeRegister(REG_PA_CONFIG, 0x70 | level);`

So: Pmax = 10.8+0.6*7 = [dBm]
And Pout = Pmax - (15-OutputPower) = 15 - (15-OutputPower) and you want 10. So you set OutputPower to 10 (15 - (15-10)) = 10.

`LoRa.setTxPower(10, 0);`

ChangeD20, 2021-12-18:
Thanks a lot for your help. Based on your information I could comprehend most of what to do and where to find the information in the data sheet.. :-) it took me some time, I find it really complicated.

Some questions are still left:

> MaxPower is set in this scenario at 7 by the library:

> `writeRegister(REG_PA_CONFIG, 0x70 | level);`

How I can read 7 out of this? And why 7 and not another MaxPower?

> And Pout = Pmax - (15-OutputPower) = 15 - (15-OutputPower) and you want 10. So you set OutputPower to 10 (15 - (15-10)) = 10.

Equivalent to this the calculation for PA_Boost pin would be:

Pout = 17 - (15 - OutputPower) = 17 - (15 - 10) = 12 , correct?

-> `LoRa.setTxPower(12, 1);` ?

Both Pins are connected to the external antenna connector and so it doesn´t matter which one I take when setting up TxPower, right?

Whats about the use of the external spring antenna? I assume it´s gain is 2 or 3 dBi.

Do I have to consider it in OutputPower, maybe like this (like calculating wifi transmitter power):

OutputPower = Allowed Output - Antenna gain = 10 ERP dBm - 3 dBi = 7

insert in Pout_RFO_Pin = 15 dbm - ( 15 - 7) = 7

-> `LoRa.setTxPower(7, 0);`

?

Kongduino, 2021-12-19:
First of all, the reason I chose RFO pin is because it **limits** output, which is what you are trying to do, whereas the other pin is called **BOOST** for a reason :-)

> > MaxPower is set in this scenario at 7 by the library:
> > 
> > writeRegister(REG_PA_CONFIG, 0x70 | level);
> 
> How I can read 7 out of this? And why 7 and not another MaxPower?

The value for `REG_PA_CONFIG` is `0x70`. See the `7` in there...? :-) That's what it is. The `0x70` value can be analysed as this:

0b0....... PA_OUTPUT_RFO_PIN
0b.111.... MaxPower
0b....0000 base for OutputPower
So you add OutputPower to your 0x70 value and get the proper value for the register. As for why 7 for the MaxPower value, no idea, but I would do that to if I was the designer of this library – most people try to get **more** juice out of their LoRa module, rather than less.

> Pout = 17 - (15 - OutputPower) = 17 - (15 - 10) = 12 , correct?
> -> LoRa.setTxPower(12, 1); ?

No, it should be `LoRa.setTxPower(12, 1);` Remember you pass `OutputPower` as the argument, not the calculation result.

Well since you want `10`, OutputPower should be 8: 17 - (15 - 8) = 10. So `LoRa.setTxPower(8, 1);`
But as I said, since you want to limit, use `PA_OUTPUT_RFO_PIN`.

> LoRa.setTxPower(7, 0);

I wouldn't bother with a spring antenna. Its effect is minimal. Also, remember that the Tx power is also limited by default by another register RegOcp, overload current protection register 0x0B, at 100 mA. So you should be well within guidelines (and have a severely limited LoRa module, by the way). Of course I remember you want ranges within 100 meters, so this should not be a problem...

ChangeD20, 2021-12-19:
thanks a lot for your help!

closed 2021-12-19 by ChangeD20