sandeepmistry/arduino-LoRa #542: setTxPower seems to have no effect

Henry-Pease, 2021-12-29:
Both with measuring the output on my laptop with an SDR dongle and by looking at the RSSI from another module, I see no difference between setTxPower(17) and setTxPower(20), or even setting it at 0, 1, or 2 -- no drop in power. Adding the second parameter for PA_BOOST pin also doesn't seem to make a difference, although when I specify PA_OUTPUT_RFO_PIN I do see the power drop dramatically.

I'm using a custom PCB that's basically just a copy of the sparkfun ESP32 LoRa board, which uses the RFM95 LoRa module. Any idea if the setTxPower function is actually implemented / should be working differently than what I'm seeing?

Thanks!

IoTThinks, 2021-12-30:
Last time, I also tried to do the same thing.
Normally, a board may support either RFO or PA_BOOST but not both.

I tried the PA_BOOST, and set TX_Power also. RFO doesnt seem available for my board so there is a huge drop of RSSI.

When I changed TX_Power, the RSSI did change too but not much.
You can try to disable AGC and measure again. AGC may affect RSSI and SNR.

tausday, 2022-01-09:
Hello i have see the same. PLS Check the Register REG_PA_DAC its 0x4d.
U can check it with:

`LoRa.dumpRegisters(Serial);`

If it's 0x84 the boost is not enabled
If it's 0x87 the boost is enabled

I have checked at my Heltec at is was x84, so i have add
`LoRa.setTxPower(20,PA_OUTPUT_PA_BOOST_PIN);` After `LoRa.begin(868E6)` was successfully loaded.

Now i have 0x87 and the RSSI was dropped from -86dBmm to -79dBmm

Give it a Try

IoTThinks, 2022-01-10:
@tausday By right, when setting setTxPower, REG_PA_DAC should already be either 0x87 or 0x84.
https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.cpp#L447