sandeepmistry/arduino-LoRa #543: Can LoRa Receiver wake Arduino up from sleep?

hoseinaghajari, 2022-01-07:
Hi, I have an _Arduino UNO_ and _LORA RA-02_ module. The device works with a _Battery_ so Due to have _low power consumption_ I have to set Arduino in sleep mode: also I want to use external interrupt for waking Arduino up from sleep mode, so I want to attached DIO_0 pin of LORA to D2(int0) of Arduino. 
**But my question is when a packet receive, can LORA trigger the INT0 through the DIO_0 pin while Arduino is in sleep and wake it up? can I receive a packet when Arduino(not LORA) is in sleep?**

Kongduino, 2022-01-08:
First keep in mind that the Arduino Uno is 5 volts, and the Ra-02 is 3.3 volts. Be careful which pins you use. Second, yes you can set up your Uno to go to sleep, and wake up with `attachInterrupt (interrupt, ISR, mode);`. Something like this:

```c
/**
   Author: Ab Kurk
   version: 1.0
   date: 24/01/2018
   Description:
   This sketch is part of the beginner's guide to putting your Arduino to sleep
   tutorial. It is to demonstrate how to put your Arduino into deep sleep and
   how to wake it up.
   Link To Tutorial http://www.thearduinomakerman.info/blog/2018/1/24/guide-to-arduino-sleep-mode
*/

#include <avr/sleep.h> // This AVR library contains the methods that controls the sleep modes
#define interruptPin 2 // Pin we are going to use to wake up the Arduino

void setup() {
  Serial.begin(115200); // Start Serial Comunication
  pinMode(LED_BUILTIN, OUTPUT); // We use the LED on pin 13 to indicate when Arduino is asleep
  pinMode(interruptPin, INPUT_PULLUP); // Set pin d2 to input using the buildin pullup resistor
  digitalWrite(LED_BUILTIN, HIGH); // TurnLED on
}

void loop() {
  delay(5000); // Wait 5 seconds before going to sleep
  Going_To_Sleep();
}

void Going_To_Sleep() {
  sleep_enable(); // Enable sleep mode
  attachInterrupt(0, wakeUp, LOW); // Attach an interrupt to pin D2
  set_sleep_mode(SLEEP_MODE_PWR_DOWN); // Set the sleep mode, in our case full sleep
  digitalWrite(LED_BUILTIN, LOW); // Turn LED off
  delay(1000); //wait a second to allow the led to be turned off before going to sleep
  sleep_cpu(); //activating sleep mode
  Serial.println("just woke up!"); // The next line of code is executed after the interrupt
  digitalWrite(LED_BUILTIN, HIGH); // Turn LED on
}

void wakeUp() {
  Serial.println("Interrrupt Fired"); // Print message to serial monitor
  sleep_disable(); // Disable sleep mode
  detachInterrupt(0); // Remove the interrupt from pin 2;
}
```

In the `wakeUp` function you'll have to  put your own code to read the packet, etc. And in `setup` you'll have to initialize the LoRa module as usual.

hoseinaghajari, 2022-01-12:
> First keep in mind that the Arduino Uno is 5 volts, and the Ra-02 is 3.3 volts. Be careful which pins you use. Second, yes you can set up your Uno to go to sleep, and wake up with `attachInterrupt [...]

No, I mean LoRa wake up Arduino. what exactly wakes Arduino up? I've asked that Can LoRa wake Arduino from sleep mode?

Kongduino, 2022-01-12:
That was your question, and I answered it...

hoseinaghajari, 2022-02-26:
Regarding to my research, I've found that LoRa Can't wake Arduino up from sleep.

closed 2022-02-26 by hoseinaghajari