sandeepmistry/arduino-LoRa #545: Problem recieving with ESP32 and SX1278

Nehuix, 2022-01-12:
Hi folks! I am currently having a weird issue while trying the LoraSender and LoraReciever examples.
I am trying to communicate an ESP32 and a Arduino Nano via LoRa using two SX1278. It turns out that when I use the ESP32 as a transmitter and the Nano as a reciever I have no problems, the Nano recieves the message with no further issues, but when I reverse the system (use ESP32 as a reciever and Nano as a transmitter) using the exact same codes (just changing the ".setPins()" of course) the Nano transmits with no problem, but the ESP32 doesn't recieve anything at all. Also the serial por doesn't show the "Starting LoRa failed!" message and I even wrote a line that prints a message if the Initialized properly.
I hope you guys can give me a hand.
Thanks in advance!

These are the codes I use:

**Reciever:**

#include <LoRa.h>
#include <SPI.h>
 
#define ss 5
#define rst 14
#define dio0 2

//arduino
//#define ss 10
//#define rst 9
//#define dio0 2

void setup() 
{
  Serial.begin(115200);
  while (!Serial);
  Serial.println("LoRa Receiver");
 
  LoRa.setPins(ss, rst, dio0);    //setup LoRa transceiver module

  while (!LoRa.begin(433E6))     //433E6 - Asia, 866E6 - Europe, 915E6 - North America
  {
    Serial.println("noanda.");
    delay(500);
  }
  //LoRa.setSyncWord(0xA5);
  Serial.println("LoRa Initializing OK!");
}
 
void loop() 
{
  
  int packetSize = LoRa.parsePacket();    // try to parse packet
  if (packetSize) 
  {
    
    Serial.print("Received packet '");
 
    while (LoRa.available())              // read packet
    {
      String LoRaData = LoRa.readString();
      Serial.print(LoRaData); 
    }
    Serial.print("' with RSSI ");         // print RSSI of packet
    Serial.println(LoRa.packetRssi());
  }
}

**Sender**

#include <LoRa.h>
#include <SPI.h>
 
int counter = 0;

//#define ss 5
//#define rst 14
//#define dio0 2

//arduino
#define ss 10
#define rst 9
#define dio0 2

void setup() {
  Serial.begin(115200);
  while (!Serial);

  Serial.println("LoRa Sender");
  LoRa.setPins(ss, rst, dio0);

  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
  Serial.print("Sending packet: ");
  Serial.println(counter);

  // send packet
  LoRa.beginPacket();
  LoRa.print("hello ");
  LoRa.print(counter);
  LoRa.endPacket();

  counter++;

  delay(1000);
}

dvio, 2022-01-12:
How did you power the lora module on the nano side? If you powered it from the 3.3v pin it may be underpowered, try using an external power for the lora module

Nehuix, 2022-01-13:
> How did you power the lora module on the nano side? If you powered it from the 3.3v pin it may be underpowered, try using an external power for the lora module

I can't thank you enought! You just saved me!. That was the solution. Thank you very much!

closed 2022-01-13 by Nehuix