sandeepmistry/arduino-LoRa #548: Is it possible to use 2 LoRa (sx1278) modules at the same time? (with one mcu)

misaalanshori, 2022-01-23:
Does this library work with 2 different Lora modules at the same time with one MCU? I'm planning on getting another 2 modules so the board can communicate simultaneously.

benner, 2022-01-23:
You should be able to initiate two `LoRaClass` objects and set different pins for each with `LoRaClass::setPins`

Kongduino, 2022-01-23:
> Does this library work with 2 different Lora modules at the same time with one MCU?

Yes it can. You will need different pins for `LoRa.setPins(SS, RST, DI0);` while keeping the same pins for MOSI/MISO. However the second part of your question brings doubt over the purpose of your question:

> I'm planning on getting another 2 modules so the board can communicate simultaneously.

If one module is receiving, you'll have to deal with the reception, and unless you MCU has at least two cores, and you know how to handles multiple tasks, how are you planning to simultaneously send stuff if your MCU is busy? One module per board will be more than enough,

misaalanshori, 2022-01-24:
Yeah, I was thinking of using an ESP32 (instead of ESP8266) because it has more pins and also more cores. But I ended up just looking for a different solution mostly because of cost and also because it looks like adding another module actually might just add more problems.

closed 2022-01-24 by misaalanshori