sandeepmistry/arduino-LoRa #554: Starting LoRa failed

pramasenja, 2022-01-31:
Hi, i failed when tried to starting the LoRa. I tried using the demo code and using LoRa RFM95W with arduino mega 2560. I've used Logic level converter but it still failed to start.

And then i add `LoRa.setSPIFrequency(1E6)` and still failed to start. (Already tried from `8E6` to `1E6`)

And then i add `LoRa.setPins(10, 9, 2)` and still failed to start

Here is some pictures of my wiring.
[image]
[image]
[image]

[image]
 
Here is my code :
```
#include <SPI.h>
#include <LoRa.h>

int counter = 0;

void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Sender");
  LoRa.setPins(10, 9, 2);
  LoRa.setSPIFrequency(1E6);
  if (!LoRa.begin(915E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
  Serial.print("Sending packet: ");
  Serial.println(counter);

  // send packet
  LoRa.beginPacket();
  LoRa.print("hello ");
  LoRa.print(counter);
  LoRa.endPacket();

  counter++;

  delay(5000);
}
```
Sorry, I'm still a beginner at this.

Kongduino, 2022-02-01:
It's hard to judge from the photos, but it looks like pins 9 and 10 have nothing connected to them.

pramasenja, 2022-02-02:
Hi @Kongduino , Thanks for the help, but it still failed to start. (Starting LoRa Failed)

Sorry if my photos is not clear enough, this is the wiring schema picture : 
[image]

Sorry if the logic level converter image is different from the photo, but the pins is still the same.

And here is the current photos of lora and arduino :
[image]
[image]
[image]

Kongduino, 2022-02-02:
The first issue I can see is that HV is connected to 3v3, whereas it should be connected to 5v. You're using a bidirectional logic converter for a reason: to convert between 5v and 3.3v... So on the HV side you need to give them a reference voltage that matches the level of the various pins, ie 5v.

pramasenja, 2022-02-02:
Thanks for the explanation @Kongduino.
Should i use external power supply 5v?
i tried to move the pin to 5v, and it still failed.
[image]
[image]

EsKliwoN, 2022-02-03:
i think i know what the problem is. The problem is in your Logic Converter.
I have same experience with that logic converter and my LoRa can not run properly. And i just realize that my Logic Converter doesn't support high Frequency for SPI communication that used by LoRa (MISO, MOSI, NSS, CLK).  Problem Solving :
1. Change your board with 3.3V working voltage like esp32 or STM32
2.  Try to change your logic converter with IC inside them. Like TXS0108E 8 Channel ( https://components101.com/modules/txs0108e-bi-directional-logic-level-converter-module )

pramasenja, 2022-02-04:
Thanks @EsKliwoN , I will try to buy a board with 3.3v next week

pramasenja, 2022-02-17:
Thank you @EsKliwoN , esp32 works fine.
I think the problem is in my Logic Converter.
I'm gonna close this issue

[image]
[image]

closed 2022-02-17 by pramasenja