sandeepmistry/arduino-LoRa #557: RFM95W (SX1276) G0/IRQ pin not working (maybe RegIrqFlags uninitialized?) -> LoRa Receiver Callback not working (on NANO 33 BLE)

ratio-x, 2022-02-02:
There seems to be a problem with RegIrqFlags, which causes problems in **LoRaSimpleGateway** and **LoRaReceiverCallback** example sketches. The following is far beyond my knowledge, so please bear with me if anything has not been described quite correctly.

I have 2 identical hardware setups on breadboard, working with Arduino IDE 2.0.0-rc3

NANO 33 BLE (powered by USB) <--> Adafruit RFM95W breakout (with SMA antenna)
D13	<--> SCK
+3V3 <--> VIN
GND <--> GND
D12	<--> MISO
D11 <--> MOSI
D4 <--> CS
D3 <--> G0/IRQ
D2 <--> RST

Power supply is sufficient, according to NANO's datasheet there should be 950 mA. All digital pins of NANO are interrupt capable.
In the example sketches I'm working with following customization:

```
const long frequency = 868E6;  // LoRa Frequency

const int csPin = 4;          // LoRa radio chip select
const int resetPin = 2;        // LoRa radio reset
const int irqPin = 3;          // change for your board; must be a hardware interrupt pin

LoRa.setPins(csPin, resetPin, irqPin);
```

I was able to start LoRa communication with the minimalistic **LoRaSender** and **LoRaReceiver** sketches, not caring about interrupts and callbacks. So this works with or without **D3 <--> G0/IRQ** connected. Output as expected:

```
LoRa Receiver
Sending packet: 9
Received packet 'hello 9' with RSSI -25
Sending packet: 10
Received packet 'hello 10' with RSSI -31
Sending packet: 11
Received packet 'hello 11' with RSSI -31
Sending packet: 12
Received packet 'hello 12' with RSSI -31
Sending packet: 13
Received packet 'hello 13' with RSSI -31
```

Using **LoRaSimpleGateway** and **LoRaReceiverCallback** sketches, I'm running into problems. First try was with **LoRaSimpleGateway**. When RFM95W's G0/IRQ pin is connected to NANO's D3 pin, then the Serial Monitor is freezing after exactly one packet was sent outward from gateway. This gateway freeze happens in any case - without node transmissions (powered off) and with node transmissions (working properly without freeze).

```
LoRa init succeeded.

LoRa Simple Gateway
Only receive messages from nodes
Tx: invertIQ enable
Rx: invertIQ disable

Send Message!   ---> freeze!
```

When disconnecting **D3 <--> G0/IRQ** the gateway does not stop working after first packet sent, again independent from node transmissions.

```
LoRa init succeeded.

LoRa Simple Gateway
Only receive messages from nodes
Tx: invertIQ enable
Rx: invertIQ disable

Send Message!
Send Message!
Send Message!
Send Message!
Send Message!
[...]
```

Next try was with **LoRaReceiverCallback** (just receiving, not sending) in combination with (proven working) **LoRaSender**. Unfortunately no confirmed readings:

```
LoRa Receiver Callback
```

For further investigation, if the Receiver received a packet at all, I checked RFM95W's G0/IRQ pin (disconnected from NANO's D3) with a logic analyzer. Powering the receiver **after** starting logic analyzer recording makes visible, that G0/IRQ pin goes from **LOW** to **HIGH** immediately after powering.

Inserting a delay of 10 seconds into the setup function of the Receiver (**LoRaReceiverCallback**) makes visible, that G0/IRQ pin goes from **LOW** to **HIGH** 10 seconds after powering.

```
  Serial.println("LoRa Receiver Callback");
  
  if (!LoRa.begin(frequency)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
  
  delay(10000);   // ----------- 10 seconds wait -----------------

  // Uncomment the next line to disable the default AGC and set LNA gain, values between 1 - 6 are supported
  // LoRa.setGain(6);
  
  // register the receive callback
  LoRa.onReceive(onReceive);

  // put the radio into receive mode
  LoRa.receive();
```

I repeated this WITHOUT Sender powered and noticed, that that G0/IRQ pin remains **LOW**.

I investigated with the **LoRaSimpleGateway** sketch, RFM95W's G0/IRQ pin not connected to NANO's D3 pinconnected, but connected to the logic analyzer. G0/IRQ pin goes from **LOW* to **HIGH** with (or after) first transmission. Repeatingly after 3 seconds (during each transmission) it goes to **LOW** for 74 ms (milliseconds) and back again to **HIGH**.

**LoRaDumpRegisters** with my (above mentioned) customized pins and frequency shows following output:

```
LoRa Dump Registers
0x0: 0x48
0x1: 0x81
0x2: 0x1A
0x3: 0xB
0x4: 0x0
0x5: 0x52
0x6: 0xD9
0x7: 0x0
0x8: 0x0
0x9: 0x8F
0xA: 0x9
0xB: 0x2B
0xC: 0x23
0xD: 0x1
0xE: 0x0
0xF: 0x0
0x10: 0x0
0x11: 0x0
0x12: 0x0
0x13: 0x0
0x14: 0x0
0x15: 0x0
0x16: 0x0
0x17: 0x0
0x18: 0x10
0x19: 0x0
0x1A: 0x0
0x1B: 0x0
0x1C: 0x0
0x1D: 0x72
0x1E: 0x70
0x1F: 0x64
0x20: 0x0
0x21: 0x8
0x22: 0x1
0x23: 0xFF
0x24: 0x0
0x25: 0x0
0x26: 0x4
0x27: 0x0
0x28: 0x0
0x29: 0x0
0x2A: 0x0
0x2B: 0x0
0x2C: 0x0
0x2D: 0x50
0x2E: 0x14
0x2F: 0x45
0x30: 0x55
0x31: 0xC3
0x32: 0x5
0x33: 0x27
0x34: 0x1C
0x35: 0xA
0x36: 0x3
0x37: 0xA
0x38: 0x42
0x39: 0x12
0x3A: 0x49
0x3B: 0x1D
0x3C: 0x0
0x3D: 0xAF
0x3E: 0x0
0x3F: 0x0
0x40: 0x0
0x41: 0x0
0x42: 0x12
0x43: 0x24
0x44: 0x2D
0x45: 0x0
0x46: 0x3
0x47: 0x0
0x48: 0x4
0x49: 0x23
0x4A: 0x0
0x4B: 0x9
0x4C: 0x5
0x4D: 0x84
0x4E: 0x32
0x4F: 0x2B
0x50: 0x14
0x51: 0x0
0x52: 0x0
0x53: 0x10
0x54: 0x0
0x55: 0x0
0x56: 0x0
0x57: 0xF
0x58: 0xE0
0x59: 0x0
0x5A: 0xC
0x5B: 0x0
0x5C: 0x7
0x5D: 0x0
0x5E: 0x5C
0x5F: 0x78
0x60: 0x0
0x61: 0x1C
0x62: 0xE
0x63: 0x5B
0x64: 0xCC
0x65: 0x0
0x66: 0x1
0x67: 0x50
0x68: 0x0
0x69: 0x0
0x6A: 0x0
0x6B: 0x0
0x6C: 0x0
0x6D: 0x0
0x6E: 0x0
0x6F: 0xB
0x70: 0xD0
0x71: 0x0
0x72: 0x12
0x73: 0x0
0x74: 0x0
0x75: 0x0
0x76: 0x0
0x77: 0x0
0x78: 0x0
0x79: 0x0
0x7A: 0x0
0x7B: 0x0
0x7C: 0x0
0x7D: 0x0
0x7E: 0x0
0x7F: 0x0
```

I continued reading SX1276's datasheet and cource code of LoRa.cpp, searching for the code, which initializes RegIrqFlags (0x12), leading to **handleDio0Rise()** function and **writeRegister(REG_IRQ_FLAGS, ...)**. This is the point where my skills are limited.

I suspect something is wrong with it ( RFM95W does not give interrupt signals, while for very sure it receives packets, so I suspect it was not properly initialized by LoRaReceiverCallback), but I can't prove it.

Any ideas?

a71cip, 2024-06-24:
I think i have the same problem with 2 circuits (Arduino Pro Mini + RFM95 Lora Module).
LoRaSender and LoRaReceiver sketches works fine but LoRaReceiverCallback does not.
I'm using the default SS = pin 10, RESET = pin 9, DIO0 = pin2 and connections are made in the same way on the circuits.
According to datasheets on Arduino Pro Mini pin 2 is interrupt capable.

Did you find any solution about this problem ?

ratio-x, 2024-06-25:
No, I gave up the RFM95W (SX1276) module and switched to the EBYTE E220-900T30D (LLCC68), which works fine for me.

morgan-wild, 2025-03-12:
I have the same problem with SW1278 chips. 

Does the LLCC68 is directly usable with this LoRa library (sandeepmistry/arduino-LoRa) without changes?