sandeepmistry/arduino-LoRa #558: Poor RSSI - Field sight 400m -120 dbi, 1 m -60 dbi at TXpower 20

StevenArduino1, 2022-02-03:
I am using a Hope RFM95 (915 Mhz), spread factor 10, default bandwidth, module soldered onto a ESP8266 plate with resistors removed and SMA connector soldered onto the plate.

 I read other peoples Lora tests and they get about  RSSI - 35 dbi at 1m and around RSSI - 100 dbi at 400 m line of sight.  I am getting around - 120 dbi at about 450 m and I am getting occasional data corruption, I need to get to 800m and  I am having issues at 450 m . I am using 12 bdi base antenna and 5 dbi node antenna, I have tried many other antennas and putting the 12 dbi antenna at the base gave me about 6 dbi improvement, but still not good.

I believe I should be getting around -100 dbi at about 450m and there is a problem. I am suspicious on some sort of antenna or TX power setting issue.  In my office at best (LoRa.setTxPower(20, PA_OUTPUT_PA_BOOST_PIN);) using helix antennas at 1 m apart I get RSSI - 60 dbi.  Wether I use LoRa.setTxPower(20, PA_OUTPUT_PA_BOOST_PIN); or LoRa.setTxPower(20); makes no difference. At 1 m at TXpower 2 RSSI is -72 dbi,  TXpower 17 RSSI is -64 dbi and LoRa.setTxPower(14, PA_OUTPUT_RFO_PIN); RSSI - 57 dbi . I get better results using RFO_PIN rather than default PA_BOOST???? 

I have read the posts in Poor RSSI #47 HamedJafarzadeh commented he " connecting FEM_CTX to RXTX pin on my module, increased RSSI significantly. Now I'm getting around ~35." I have no idea how he did it, can someone explain. Or any other suggestions?

Kongduino, 2022-02-03:
Alright, as someone who achieves 10 km between modules on a regular basis, these figures kind of worry me. Crappy antennas don't excuse/explain everything – even with old TTGO v 1.0 and helix antennas I got 2 km in a busy urban environment. These figures do remind me of the crappy M5Stack Ra-01/Ra-02 modules. I never was able to get anything better than 800 meters with them.

First, the SF/BW settings are important. You say `spread factor 10, default bandwidth` – default for whom? If that's from the datasheet, probably 7? Second, these settings are maybe not the ideal ones for long-range. The higher the SF, the lower the BW, the better.

Here are settings with which I tend to initialize my SX127x modules. Note that in the library in moved `writeRegister` and `readRegister` to `public` so I could do this.

```c
  // BW = 6: 62.5 kHz, CR = 1: 4/5, HM = 0
  uint8_t reg1 = 0x62;
  // SF = 12: 12, CRC = 1
  uint8_t reg2 = 0xC4;
  // LDRO = 1, AGCAutoOn = 0-->LNA gain set by register LnaGain
  uint8_t reg3 = 0x08;
  // 7:5 LnaGain 001 -> G1 = maximum gain
  // 1:0 LnaBoostHf 11 -> Boost on, 150% LNA current
  uint8_t reglna = 0b00100011;
  LoRa.writeRegister(REG_OP_MODE, MODE_LONG_RANGE_MODE | MODE_SLEEP);
  delay(10);
  LoRa.writeRegister(REG_PA_CONFIG, 0xFF);
  LoRa.writeRegister(REG_MODEM_CONFIG_1, reg1);
  LoRa.writeRegister(REG_MODEM_CONFIG_2, reg2);
  LoRa.writeRegister(REG_MODEM_CONFIG_3, reg3);
  LoRa.writeRegister(REG_LNA, reglna);
  LoRa.writeRegister(REG_PA_DAC, PA_DAC_HIGH); // That's for the transceiver
  LoRa.writeRegister(REG_OP_MODE, MODE_LONG_RANGE_MODE | MODE_STDBY);
  delay(10);
```

Remove the line with `REG_PA_DAC` for the receiver, and you should be good to go. Unless you have hardware issues (always possible since it's a DIY board), you should see great improvements in range.

StevenArduino1, 2022-02-03:
Kongduino, thank you for taking the time to reply.  

I am using the sandeepmistry library and I I interpret default bandwidth is 125 khz, I do not believe it is a bad antenna as you suggest, I have tested many different antenna types, but I think is is something to do with the TXpower or signal power delivered to the antenna. I am using the correct antenna SMA connectors (not RP).  have not learnt to write settings to the register "LoRa.writeRegister" and I assume this is a direct way to set the module rather than using the sandeep library.  I will try your above settings. I see others have a similar problem (-50 to 60 dbi RSSI at 1 m) and there could be just one simple setting not properly working that is causing the issue. I feel there is a loss of about 15 dbi that makes all the difference in long range, when using the 20 dbi TX power lora setting I do notice the amp meter jump briefly to 150 mA, so something is taking power, I did not record the mA at lower TXpower setting to compare (I will do). I also gave the module 3.3v  power from a different source in case the ESP32 was not providing enough power and it made no difference, the ESP32 seemed to be giving enough power. 

In the above code which sets the TXpower? Can you please put a comment of what each setting is for?  I will update my results.  Are there any good guides on how to use "LoRa.writeRegister" to change settings instead of a library?  Are there other libraries I can try, I have two ESP32 and sending messages between them, one transmit and one receive.  I am keen to hear from others what type of RSSI are they achieving?

Kongduino, 2022-02-04:
> In the above code which sets the TX power?
The line setting REG_PA_CONFIG does that:

`LoRa.writeRegister(REG_PA_CONFIG, 0xFF);`

[image]

0xFF = 0b11111111

PaSelect: 1 --> PA_BOOST pin
MaxPower: 111 --> 7, equ Pmax=10.8+0.6*7 [dBm] = 15
OutputPower: 1111 --> 15, equ Pout=17-(15-OutputPower) = Pout=17-(15-15) = 17

> Can you please put a comment of what each setting is for?

Look again, the comments are there, above each register assignment. :-)

> Are there any good guides on how to use "LoRa.writeRegister" to change settings instead of a library?

None that I know of. I have written my own adjunct library, as a helper to this library. For instance, my `void displayRegisters()` function decodes the registers in plain English. Stuff like that. I studied Semtech's `SX1276 SX1277 SX1278 SX1279 Datasheet` until I knew most of it by heart, and understood the registers. That's how I came, for instance, to write a real random function (this library's version is fubared). I probably should write a guide to the registers, something a little less terse than the datasheet. I did write a GUI application that helps me set up the registers when I am too lazy to do binary arithmetic.

[image]

StevenArduino1, 2022-02-04:
Kongduino, thank you for the explanation, I will try it.

I noticed in my code I did have LoRa.setTxPower(20);  before Lora.begin, then I moved it to after and added the second argument LoRa.setTxPower(20, PA_OUTPUT_PA_BOOST_PIN); .. In my office test this did not seem to make a difference, however I need it to work better in the field.

I did field testing today.  Spread factor 10, default bandwidth 125 khz and default coding rate using sandeep library. On the base station I have a 1.1m 12 dbi antenna, When I used it instead of a 30 cm magnetic base antenna it improved the dbi at the base station on one transmitter by 8 dbi and the other by 1-2 dbi Both are same antennas.  I also tried a 90 cm 10 dbi antenna at the base and it was no better than the 30 cm antenna, then when I switched back to the 12 dbi antenna it dropped the dbi by 10 dbi.  I will continue to use the 12 dbi antenna.  Also the SNR use to be about -5 to - 15, now its about 4.  I interpret from literature that every 3 dbi equated to doubling the distance, so every dbi improvement is important.   I also notice that sometimes the dbi can change by 5 daily , perhaps humidity can affect transmission?

The antenna is on a roof about 5m above ground, the land is slightly undulating sand hills rising up and down 3 m  and 400 m apart across the plain, I live in a arid zone.  At 400 m with the base antenna visible and using a small 10 cm transmitter antenna , holding the transmitter antenna at head height (I assume it was about 3 dbi helical) I obtained RSSI - 92 dbi, SNR 7.5. I then changed to a 30 cm antenna (supposedly 7 dbi) and obtained RSSI -88 dbi and SNR 10.7.  I will continue to use the 30 cm antenna for testing. If I interpret correctly; when changing from a 3 dbi antenna to a 7 dbi, a 4 dbi improvement should be expected according to the math!

I then tested a 800 m from the base antenna, with the base antenna visible, I obtained RSSI -100 dbi and SNR 10.5. I then went 1300 m away with the base antenna behind a single wall of tall trees about 300 m from the base antenna. I obtained a RSSI of - 115 and SNR 2.75.  I work in agriculture, oranges, and my transmitters are used to transmit soil moisture and other data.  I went into the orange block with the transmitter at 2 m height, the base station is obscured by two single rows of tall trees about 100 m apart, I obtained a RSSI of -100 dbi; but then when I  changed of code on the actual logging station that is using a different antenna I got a RSSI of - 108 and SNR 5.  I will change the logger station antenna a later to see if it improves. 

The 12 dbi antenna at the base made a definite improvement.  The new code seemed to have also made a difference and now I feel I can use the logger for my application (monitor soil moisture in a orchard). I think I am getting good results (what do you think?), but some how I feel I can get 10 dbi better from the discussion by HamedJafarzadeh in the post  mentioned above.  I will also try the code kindly explained by Kongduino and then provide updates on the progress.  I am sending 96 bytes of data and I still getting occasional errors in the data (often the same wrong number presented), I am still trying to fix that issue.  Thank you again Kongduino for your post.

Kongduino, 2022-02-04:
Thanks for the detailed answer. The figures are better but I have the feeling they could get better. At least now you can LoRaize your orange grove :-)

`LoRa.begin()` sets the TX power in its code: it calls `setTxPower(17);`, so anything that happens before `LoRa.begin()` is lost.

Trees are often an impediment to RF transmission, so that could explain the lower performance. However, you might me able to counterbalance this with a unidirectional antenna if your orange trees are all in the same direction: a Yagi would work wonder. I do tests in the 433, 868 and 915 ranges with Yagis, and Gawd do they help. I live in a very urban (think Godzilla-sized buildings) area that's also very hilly. And I get transmissions a few km away, inside the elevated subway. I have switched to antennas that are very well tuned – 86x for 868, 91x for 915, no more 860-950 crap :-). They're 16 cm long, +/- a cm, and perform very well. They beat my 30-cm antennas with a wider frequency range every single time.

My Yagis are cheapos bought from Chinese resellers (I live in Hong Kong), and they're "good enough" – I tested them on my VNA, and determined which frequencies would work well for each. Instead of trying to force them to work at freq X, I use them at the freq that showed the lowest SWR... :-) It turns out that for my supposedly 433 MHz antenna, 470 (which is good for China) is actually the better frequency. So that's what I use!

Of course if you are on 915US, you probably have a narrower legal range, but it's worth a try.

Do try the code I gave you – at SF12 / BW6, plus the transmitter optimizations, you should see marked improvements. That's how I reached 2 km with crappy TTGO devices.

FathiMahdi, 2022-09-26:
Hi @Kongduino  thanks for the solution I am having an error while updating the LoRa parameters from the register
`    LoRa.begin(915E6);
    LoRa.writeRegister(REG_OP_MODE, MODE_LONG_RANGE_MODE | MODE_SLEEP); // set lora mode modulation
    delay(10);

    LoRa.setSignalBandwidth(10.4E3); // set bandwidth
    LoRa.setCodingRate4(8); // coding rate
    LoRa.setSpreadingFactor(12); // set spread factor
    LoRa.enableCrc(); // enable crc
    LoRa.writeRegister(REG_PAYLOAD_LENGTH, PayloadLength); // set the payload lenght
    LoRa.writeRegister(Reg_HopPeriod, HopPeriod); // set the payload lenght
    LoRa.writeRegister(REG_MODEM_CONFIG_3, ModemConfig3); // enable low data rate optimization and LNA gain
    LoRa.setPreambleLength(12); // set preamble length
    LoRa.writeRegister(REG_LNA, reglna);// set the sensetivity of the receiver boost the LNA current
    LoRa.writeRegister(REG_PACONFIG, PaConfig); // set max power and boost 
    delay(10);
    LoRa.writeRegister(REG_OP_MODE, MODE_LONG_RANGE_MODE | MODE_STDBY); // set lora mode modulation`

Kongduino, 2022-09-26:
Please post the full code, surrounded by three backticks each. As it is now, it's unreadable.

FathiMahdi, 2022-09-26:
Solved. I just noticed that I need to change the writeRegister method from private to public thanks.

adityaprabhu16, 2023-08-11:
@FathiMahdi I'm curious - what register is the Reg_HopPeriod, and what is generally a good PayloadLength? Additionally, should this block of code be placed before or after LoRa.begin()? Thank you!