sandeepmistry/arduino-LoRa #559: LoRaDuplex not working properly

Nehuix, 2022-02-04:
Hi folks! I'm an currently working on a project that nees a duplex communication so I tried the LoRaDuplex example to see how it works.
I am using an Arduino Nano and a ESP32. Also I make a few modifications such as not using the "setPint()" function and leaving it with the default pins for the Nano and changing it so it can be use int the ESP32. This configuration works since I used i for the LoRaReciever and LoRa Sender.
So here is my problem: I have a hard time making them communicat. First it seems like they both send the message, but none recieves it. Sometimes it works and suddenly stops doing so. I am using an external source of power such as the Breadboard 3.3/5V power supply (working at 3.3V).
Hope any of you can help me. Thanks in advance

PepeProf, 2022-02-04:
Hello!

It might be loose connections. The examples should work. By the way, does any error or backtrace appear on the serial monitor?

Kongduino, 2022-02-05:
If both modules are transmitting at the same time, they won't be able to listen to each other. We need Channel Activity Detection, which is not implemented in this library. There's [a fork](https://github.com/szotsaki/arduino-LoRa) that has it, so it might be worth having a look at that.

Call LoRa.cad(). If there's activity, wait. No activity, send.

CAD is on my todo list.