sandeepmistry/arduino-LoRa #564: [Not an issue] Lib successfully tested with EByte E32-xxxM20S

ezcGman, 2022-02-28:
Hey there,

I'm not raising an issue, but I just wanted to let you know (whoever might find this helpful) that I've successfully tested this lib with an EByte E32-900M20S (https://www.ebyte.com/en/product-view-news.html?id=1613), which also uses the Semtech SX1276. EByte has quite some modules using that chip, but they usually release modules with an own little MCU on there, so you talk over UART/Serial interface with them. But they have a few modules that expose the actual SPI interface of the SX1276.

The only little difference to the RFM95 is that the EByte also has two pins to either switch RX or TX. So whenever you want to transmit, you need to pull the TXEN pin high and whenever you want to receive, you pull the RXEN pin high.
I saw that this lib doesn't have support for these pins, because the RFM95 seems to handle that on it's own. So I was just thinking if you might want to add support for these pins? Not really required, but just asking :)

Maybe also worth to add that module to the list of supported modules?

Let me know if I can do any kind of testing or something :)

Greetings,

Andy!

lamasseriadipolverara, 2022-09-25:
Can you explain how did you do it? 
Thank you

ezcGman, 2022-09-25:
Heyhey,

good timing actually, as I had to let that project rest for some time, but I'm picking it up in the coming days. I gonna dig out some example code and post it here :)

Greetings,

Andy!

ezcGman, 2022-09-26:
So here's a stripped down snipped from a bigger piece of code I have and that works wonderful with an ESP 32 Devkit and the E32 900M20S:
```
#include <SPI.h>
#include <LoRa.h>

#define PIN_LORA_NSS 15
#define PIN_LORA_RST 16
#define PIN_LORA_DIO0 18
#define PIN_LORA_TXEN 2
#define PIN_LORA_RXEN 4

#define LORA_SYNC_WORD 0x11

bool loraInitDone = false;
const byte maxConnTries = 15;
SPIClass spiE32(HSPI);

void connectLoRa() {
  if (!loraInitDone) {
    int loraConnTries = 0;

    LoRa.setSPI(spiE32);
    LoRa.setPins(PIN_LORA_NSS, PIN_LORA_RST, PIN_LORA_DIO0);

    while (!LoRa.begin(868E6) && loraConnTries < maxConnTries) {
      loraConnTries++;
      delay(1000);
    }
  
    if (loraConnTries >= maxConnTries) {
      Serial.println("Starting LoRa failed!");

      ESP.restart();
    }

    Serial.println("Starting LoRa success!");

    loraInitDone = true;

    LoRa.setSyncWord(LORA_SYNC_WORD);
  }
}

void onLoRaReceive(int packetSize) {
  if (!loraInitDone || packetSize == 0) return;          // if there's no packet, return

  Serial.println("Incoming message!");

  String message = "";                     // payload of packet
  while (LoRa.available()) {               // can't use readString() in callback, so
    message += (char)LoRa.read();          // add bytes one by one
  }

  Serial.println("Message: " + message);
  Serial.println("RSSI: " + String(LoRa.packetRssi()));
  Serial.println("Snr: " + String(LoRa.packetSnr()));
  Serial.println();
}

void loRaSetRfMode(bool receiving = true) {
  digitalWrite(PIN_LORA_RXEN, (receiving ? HIGH : LOW));
  digitalWrite(PIN_LORA_TXEN, (receiving ? LOW : HIGH));
}
void loRaSetReceivingMode() { loRaSetRfMode(true); }
void loRaSetTransmittingMode() { loRaSetRfMode(false); }

 
void setup() {
  Serial.begin(115200);
  while (!Serial);

  pinMode(PIN_LORA_TXEN, OUTPUT);
  pinMode(PIN_LORA_RXEN, OUTPUT);

  loRaSetReceivingMode();
  connectLoRa();

  Serial.println("LoRa ready!");
}

void loop() {
  if (loraInitDone) {
    onLoRaReceive(LoRa.parsePacket());
  }
}
```
And that's the pinout / schematic of a custom PCb I did where I use that module:
[image]

Hope that's helpful :)