sandeepmistry/arduino-LoRa #565: Change the variable Lora.read()

Victorio1209, 2022-03-03:
I need to change the variable that I received by LoRa for to send to ThingsBoard, but there is a error. "invalid conversion from 'int' to 'const char*' [-fpermissive]"

Thank you!

`#include <SPI.h>
#include <LoRa.h>
#include <WiFi.h>
#include <WiFiClient.h>
#include <ThingsBoard.h>    // ThingsBoard SDK
#include <stdlib.h>
#include <stdio.h>

char ssid[] = "Mariani";
char pass[] = "yu-gi-oh";
#define TOKEN               "HqKUEKoQx6Mht14PCguC"
#define THINGSBOARD_SERVER  "demo.thingsboard.io"

// Initialize ThingsBoard client
WiFiClient espClient;
// Initialize ThingsBoard instance
ThingsBoard tb(espClient);
// the Wifi radio's status
int status = WL_IDLE_STATUS;

char *Bovinos[]= {"Bovino1", "Bovino2"};

void setup() {
  Serial.begin(9600);
  while (!Serial);

  WiFi.begin(ssid, pass);
  InitWiFi();

  Serial.println("LoRa Receiver");
  LoRa.setPins(18, 14, 26);
  if (!LoRa.begin(915E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
  // try to parse packet
  int packetSize = LoRa.parsePacket();
  if (WiFi.status() != WL_CONNECTED) {
  reconnect();
  }
   if (!tb.connected()) {
    // Connect to the ThingsBoard
    Serial.print("Connecting to: ");
    Serial.print(THINGSBOARD_SERVER);
    Serial.print(" with token ");
    Serial.println(TOKEN);
    if (!tb.connect(THINGSBOARD_SERVER, TOKEN)) {
      Serial.println("Failed to connect");
      return;
   }
  }
  if (packetSize) {
    // received a packet
    Serial.print("Received packet '");

    // read packet
    while (LoRa.available()) {
      Serial.print((char)LoRa.read());
    }
    tb.sendTelemetryFloat("Bovinos", atof(LoRa.read()));
    // print RSSI of packet
    Serial.print("' with RSSI ");
    Serial.println(LoRa.packetRssi());
  }
    tb.loop();
}

  void InitWiFi()
{
  Serial.println("Connecting to AP ...");
  // attempt to connect to WiFi network

  WiFi.begin(ssid, pass);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("Connected to AP");
}

void reconnect() {
  // Loop until we're reconnected
  status = WiFi.status();
  if ( status != WL_CONNECTED) {
    WiFi.begin(ssid, pass);
    while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
    }
    Serial.println("Connected to AP");
  }
}`

Kongduino, 2022-03-04:
Please edit your post so that your code is surrounded by THREE backticks, before the first line and after the last line. As it is now it's unreadable. I do see a few problems...

Victorio1209, 2022-03-04:
OK! Thank you

Kongduino, 2022-03-04:
Ok so now that your code is readable, I see a couple of issues. Here is your `loop()`, slightly reformatted:

```c
void loop() {
  if (WiFi.status() != WL_CONNECTED) {
    reconnect();
  }
  if (!tb.connected()) {
    // Connect to the ThingsBoard
    Serial.print("Connecting to: ");
    Serial.print(THINGSBOARD_SERVER);
    Serial.print(" with token ");
    Serial.println(TOKEN);
    if (!tb.connect(THINGSBOARD_SERVER, TOKEN)) {
      Serial.println("Failed to connect");
      return;
    }
  }
  // try to parse packet
  int packetSize = LoRa.parsePacket();
  // RMK: I brought this code down to keep it with the rest of the LoRa code:
  // Above: wifi, below: LoRa
  if (packetSize) {
    // received a packet
    Serial.print("Received packet '");
    // read packet
    while (LoRa.available()) {
      Serial.print((char)LoRa.read());
    }
    tb.sendTelemetryFloat("Bovinos", atof(LoRa.read()));
    // print RSSI of packet
    Serial.print("' with RSSI ");
    Serial.println(LoRa.packetRssi());
  }
  tb.loop();
}
```

Now the problem is that you are trying to read the same packet ***twice***:

1. `Serial.print((char)LoRa.read());`
2. `tb.sendTelemetryFloat("Bovinos", atof(LoRa.read()));`

You gotta pick one... By the time you arrive at `sendTelemetryFloat` the packet is empty... I understand that you want to print the packet, then send it. So you need to save it first. Also, `tb.sendTelemetryFloat("Bovinos", atof(LoRa.read()));` – even if you hadn't consumed the packet already – wouldn't work because `LoRa.read()` only reads one byte...

So what you need to do is save the packet in a buffer, and then print it, if you want, and then do your `sendTelemetryFloat` call.

```c
  // try to parse packet
  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    // received a packet
    Serial.print("Received packet '");
    // read packet
    char buff[32] = {0};
    // adjust size if needed
    uint8_t ix = 0;
    while (LoRa.available()) {
      buff[ix++] = LoRa.read();
    }
    Serial.print(buff);
    tb.sendTelemetryFloat("Bovinos", atof(buff));
    // print RSSI of packet
    Serial.print("' with RSSI ");
    Serial.println(LoRa.packetRssi());
  }
  tb.loop();
```

The buffer `buff` receives one character at a time, at index `ix`. When the packet is consumed, you print it, and send the `atof()` conversion.

Victorio1209, 2022-03-04:
Perfect, @Kongduino! You were saved me because this is a project for my graduate in electrical engineering. Another thing, I would like to receive the "Bovinos" too, but can I separate the packet?

```
tb.sendTelemetryFloat("Bovinos", atof(buff));
```

Kongduino, 2022-03-05:
I have no experience with ThingsBoard. You'll have to look at the API.

Victorio1209, 2022-03-05:
I understand @Kongduino, but how I can divide the "buff"? Now I am receiving this packet:
Received packet 'Bovino20.00' with RSSI -73
I need to create two variables, one with the string "Bovino" and the other with "20.00".