sandeepmistry/arduino-LoRa #572: IRQ Pin Required for receiving?

kikolobo, 2022-03-23:
Hello, I am working on a pin constraint situation and I was wondering if the IRQ pin is required for poll receiving. I am looking at this example:

https://github.com/sandeepmistry/arduino-LoRa/blob/master/examples/LoRaDuplex/LoRaDuplex.ino

But it looks like it does not require the pin. I am also seeing that putting the device in (LoRa.receive()) is not required to see if you have received packets. 

I am also wondering if LoRa is full duplex or half duplex?

Thanks for any help in advance.

IoTThinks, 2022-06-19:
OnReceive() requires DIO0.
ParsePacket does not.

LoRa sx126x and sx127x are half duplex.

kikolobo, 2022-06-29:
> OnReceive() requires DIO0. ParsePacket does not.
> 
> LoRa sx126x and sx127x are half duplex.

Thank you.

closed 2022-06-29 by kikolobo