sandeepmistry/arduino-LoRa #575: LoRa receiver not working

AnandhaLakshmanan, 2022-05-06:
I am using two LoRa sx1278 with arduino uno boards. the sender node is working perfectly but the receiver doesn't show any received messages. LoRa initialization is okay in both receiver and sender.

sender code :
#include <SPI.h>
#include <LoRa.h>

int counter = 0;

void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Sender");

  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
  Serial.print("Sending packet: ");
  Serial.println(counter);

  // send packet
  LoRa.beginPacket();
  LoRa.print("hello ");
  LoRa.print(counter);
  LoRa.endPacket();

  counter++;

  delay(5000);
}

Receiver code : 
#include <SPI.h>
#include <LoRa.h>

void setup() {
  Serial.begin(9600);
  while (!Serial);

  Serial.println("LoRa Receiver");

  if (!LoRa.begin(433E6)) {
    Serial.println("Starting LoRa failed!");
    while (1);
  }
}

void loop() {
  // try to parse packet
  int packetSize = LoRa.parsePacket();
  if (packetSize) {
    // received a packet
    Serial.print("Received packet '");

    // read packet
    while (LoRa.available()) {
      Serial.print((char)LoRa.read());
    }

    // print RSSI of packet
    Serial.print("' with RSSI ");
    Serial.println(LoRa.packetRssi());
  }
}

marius-ne, 2022-05-11:
I once had the same problem, if your sender is not connected to a serial device (i.e. your PC) the line
`while (!Serial);`
will halt your program. If you remove that it should work.

efraimwilliam, 2022-11-26:
i did project with lora back then and everything is good, and i now i tried to use it again, i can't get any data from lora sender. i tried to change the arduino uno and the problem is the same. Is there any solution?

[image]

Kongduino, 2022-11-26:
You don't set SF, BW, CR parameters, so they could be different on both devices.

efraimwilliam, 2022-12-02:
> You don't set SF, BW, CR parameters, so they could be different on both devices.

Thank you for your reply. If it's okay, can i have the example of what you meant by setting SF, BW, and CR parameters?